daniestevez/gr-satellites #122: ¿Como conectar un servidor gomspace ms100 con un radio diferente al gs100?

luisguarnizo1234, 2020-07-08:
hola Dani, actualmente estoy iniciando en este mundo de los nanosatelites y me surgió la duda de si es posible conectar un servidor de gomspace ms100 a una sdr convencional. En caso de poderse realizar habria que usar un componente adicional como un microcontrolador para ejecutar el CSP?

Gracias por tu ayuda amigo

daniestevez, 2020-07-08:
Hola Luis,
No entiendo qué tiene que ver esta pregunta con gr-satellites y no conozco muy bien el GOMspace MS100, pero parece ser un servidor rackeable Dell junto con un software de GOMspace. Por tanto, supongo que puedes hacer con el hardware lo que quieras, aunque posiblemente pierdas el soporte oficial de GOMspace. En cualquier caso, creo que esta es una pregunta que le deberías hacer a GOMspace.

luisguarnizo1234, 2020-07-08:
Dani, que pena contigo. Tienes razón, esta no es una pregunta para ti.
He utilizado gr-Satelites para decodificar algunos datos del FACSAT-1, Me ha sido muy útil. 
Muchas gracias

closed 2020-07-09 by daniestevez