daniestevez/gr-satellites #123: Installing gr-satellites

leonelce3t, 2020-07-08:
Hello!
I am installing gr-satellites latest version, apparently it is correct:

leonel@leonel-X555LB:~/gr-satellites/build$ ctest -V
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/build/DartConfiguration.tcl
Test project /home/leonel/gr-satellites/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_ao40_fec_deframer

1: Test command: /usr/bin/sh "/home/leonel/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
1: Test timeout computed to be: 10000000
1: .
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.032s
1: 
1: OK
1/2 Test #1: qa_ao40_fec_deframer .............   Passed    1.67 sec
test 2
    Start 2: qa_satyaml

2: Test command: /usr/bin/sh "/home/leonel/gr-satellites/build/python/satyaml/qa_satyaml_test.sh"
2: Test timeout computed to be: 10000000
2: .
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.276s
2: 
2: OK
2/2 Test #2: qa_satyaml .......................   Passed    0.38 sec

100% tests passed, 0 tests failed out of 2

Total Test time (real) =   2.06 sec

**But now I have this problem when I use (Satellite decoder and Hexadump sink):**

Generating: '/home/leonel/untitled.py'
Generating: '/home/leonel/untitled.py'
Executing: /usr/bin/python3 -u /home/leonel/untitled.py
Traceback (most recent call last):
  File "/home/leonel/untitled.py", line 185, in <module>
    main()
  File "/home/leonel/untitled.py", line 163, in main
    tb = top_block_cls()
  File "/home/leonel/untitled.py", line 79, in __init__
    self.satellites_satellite_decoder_0 = satellites.core.gr_satellites_flowgraph(name = 'BeliefSat-1', samp_rate = samp_rate, grc_block = True, iq = False)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 91, in __init__
    satyaml = self.open_satyaml(file, name, norad)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 204, in open_satyaml
    satyaml = yamlfiles.search_name(name)
  File "/usr/local/lib/python3/dist-packages/satellites/satyaml/satyaml.py", line 128, in search_name
    raise ValueError('satellite not found')
ValueError: satellite not found

>>> Done (return code 1)

daniestevez, 2020-07-09:
Hi Leonel,

You are trying to use `BelifSat-1` as the satellite, but support for `BeliefSat-1` is only available in the `beliefsat` branch.

leonelce3t, 2020-07-09:
That is to say, that the [tutorial](https://www.youtube.com/watch?v=TceMth67r9c&t=322s) cannot be used with only the installation of gr-satellites?

Also I have had problems with the installation of dependencies and libraries in ubuntu 20.04, I have not found how to install liborc-0.4-dev 

Thanks for your help.

daniestevez, 2020-07-09:
That presentation was directed mainly towards the students building BeliefSat-1, and so used BeliefSat-1 as an example. BeliefSat-1 is not currently available in the stable version of gr-satellites (which is v3.0.0), but rather in the `beliefsat` branch. The general principles shown in that presentation can be followed with any other satellite, however.

liborc-0.4-dev in Ubuntu 20.04 is [just where it should be](https://packages.ubuntu.com/focal/liborc-0.4-dev), so I wonder where have you been looking for.

leonelce3t, 2020-07-09:
Sorry if it's obvious, but how do I install it? I already downloaded the compressed file

daniestevez, 2020-07-09:
I believe you can find plenty of information in Google about how to install packages in Ubuntu. liborc-0.4-dev is just like any other package bundled with the distribution.

But now that you have downloaded the `.deb` file, you may install it with `dpkg -i`.

leonelce3t, 2020-07-09:
Thanks for your help Dani

I can help you by sharing the problems and solutions by installing gr-satellites ubuntu 20.04, for others who want to do it, where can I put it?

daniestevez, 2020-07-10:
There is a [wiki](https://github.com/daniestevez/gr-satellites/wiki) which is somewhat un-updated and which could benefit from some work.

closed 2020-07-10 by daniestevez

N6RFM, 2020-07-11:
Hi Leonel,  

Happy to help.  Please feel free to add to the wiki, or create a new page for 20.04.  If you need help doing this, let me know and I will update.  I also believe W2RTV has also been successful using 20.04.  

Best,  
Bob N6RFM