daniestevez/gr-satellites #125: Add support for BY70-1

daniestevez, 2020-07-09:
Support for BY70-1 in [gr-lilacsat](https://github.com/bg2bhc/gr-lilacsat/) has been added a few days ago in [this commit](https://github.com/bg2bhc/gr-lilacsat/commit/c35b5d42538d4082cd00a1accc331fa2edca6e27). I should determine what needs to be added to gr-satellites in order to support this satellite.

The [flowgraph](https://github.com/bg2bhc/gr-lilacsat/blob/c35b5d42538d4082cd00a1accc331fa2edca6e27/examples/BY70-2/demod_by70-2.grc) in gr-lilacsat is the same as the one for LilacSat-1 except for a `dslpw_tm_header_parser` connected to the `out` port of the `lilacsat_lilacsat1_frame_depack` instead of going directly to a Message Debug PDU block as in LilacSat-1.

So it seems that (slightly modified) CCSDS TM Primary Headers, and probably CCSDS Space Packet Primary headers are used, but I need to check this with Wei Mingchuan or with a sample recording. It also seems that probably the frame structure is exactly the same as for Taurus-1, so people can try to use the Taurus-1 decoder with BY70-2 and report if it works.

Finally, I have seen that the Taurus-1 telemetry parser in gr-satellites doesn't have [much detail](https://github.com/daniestevez/gr-satellites/blob/3f2253ea9e69055b90a852d330bb35aac1f2fe15/python/telemetry/by70_1.py#L408) for the CCSDS header, so this should be added for BY70-2.

daniestevez, 2020-07-10:
The implementation is now done, thanks to a detailed email from Wei Mingchuan describing the telemetry format.

closed 2020-07-10 by daniestevez

ik8ozv, 2020-07-10:
Dani please what to do to add the BY70-2?????
Thanks
73's de IK8OZV

daniestevez, 2020-07-10:
It is already available in the master branch.

ik8ozv, 2020-07-10:
Apologise Dani but I see just the BY70-1. Should I have to do something to see it?
Thanks
73’s de Enzo IK8OZV

commit be294b0e (2020-07-10): Add SatYAML for BY02

This closes #125