daniestevez/gr-satellites #127: How to know the version

ik8ozv, 2020-07-11:
Hi folks,
two questions.
I have seen that Daniel has updated his gr-satellites.
Please how to update? should i need to delete the actual folder and use as before like the git clone command 
then cd gr-satellites, mkdir build, cd build etc etc etc?

The second question is how to know the actual version of gr-satellites i have installed?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-11:
The gr-satellites version that is currently installed can be checked with

```
python3 -c "import satellites; print(satellites.__version__)"
```

To upgrade the version, it is possible to install it over the older version without needing to remove the older version.

The use of the source code packages from [Github releases](https://github.com/daniestevez/gr-satellites/releases/) is recommended. So just go over to [v3.1.0](https://github.com/daniestevez/gr-satellites/releases/tag/v3.1.0), download the source code package, build it and install it over the older version.

ik8ozv, 2020-07-11:
Dani i have tried to give the command:

python3 -c "import satellites; print(satellites.__version__)"

While in the terminal window but instead to get the gr-satellites version, i get:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'satellites'

What am i doing wrong?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-19:
I have added in 2357abdaa4376b3f3ea30e0d042aec4e08fbb5a6 a command line option to print the version in `gr_satellites`, so I'm closing this issue.

closed 2020-07-19 by daniestevez