daniestevez/gr-satellites #129: Satellite Recordings

ik8ozv, 2020-07-11:
Hello,
today i was trying the new gr-satellites and i have tried the command Dani suggested me to test the flowgraphs.
i have tried with the AO-73 which was the one i had problems.
I gave the command:

gr_satellites FUNcube-1 --wavfile gr-satellites/satellite-recordings/ao73.wav --samp_rate 48e3

but got the following error:

gr-satellites/satellite-recordings/ao73.wav: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 169, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 155, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 88, in __init__
    self.setup_input()
  File "/usr/local/bin/gr_satellites", line 99, in setup_input
    return self.setup_wavfile_input()
  File "/usr/local/bin/gr_satellites", line 112, in setup_wavfile_input
    self.wavfile_source = blocks.wavfile_source(self.options.wavfile, False)
  File "/usr/local/lib/python3/dist-packages/gnuradio/blocks/blocks_swig1.py", line 404, in make
    return _blocks_swig1.wavfile_source_make(filename, repeat)
RuntimeError: can't open file

As it says at the beginner of the error :"No such file or directory" i have tried to have a look and
i went in Home/gr-satellites/satellite-recordings and i found that the folder is empty.
Why this?
Have i done something wrong?
Please what to do?
Thanks
73's de Enzo IK8OZV

K4KDR, 2020-07-11:
Please look carefully at

`https://github.com/daniestevez/gr-satellites`

... and note the 'satellite-recordings' folder is a link to another repository:

`https://github.com/daniestevez/satellite-recordings/tree/164ea835ca06d1b4e2cef546a16d9dd7ac63e437`

... from your HOME directory:

`git clone https://github.com/daniestevez/satellite-recordings.git`

... and you can then reference the following folder when referring to a recording:

`~/satellite-recordings/`

ik8ozv, 2020-07-11:
Thanks Scott,
i have done the command

git clone https://github.com/daniestevez/satellite-recordings.git

but at the end the gr-satellites/satellite- recordings folder is always empty, and have 
also find another satellite-recordings in the Home folder, so if i give the command i said the first time i get
always the same error as the folder is empty.
Do i need to pass the wav files manually from \home\satellite-recordings to \home\gr-satellites\satellite-recordings???
Thanks
73's de Enzo IK8OZV

K4KDR, 2020-07-11:
**>>>  so if i give the command i said the first time i get always the same error as the folder is empty.**

You must use the correct folder location for the recordings:

`gr_satellites FUNcube-1 --wavfile ~/satellite-recordings/ao73.wav --samp_rate 48e3`

ik8ozv, 2020-07-12:
Please i get the same ptoblem for the Nayif-1
I give the command:

gr_satellites Nayif-1 --wavfile ~/satellite-recordings/nayif-1.wav --samp_rate 48e3

but get the following error:

/home/enzo/satellite-recordings/nayif-1.wav: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 169, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 155, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 88, in __init__
    self.setup_input()
  File "/usr/local/bin/gr_satellites", line 99, in setup_input
    return self.setup_wavfile_input()
  File "/usr/local/bin/gr_satellites", line 112, in setup_wavfile_input
    self.wavfile_source = blocks.wavfile_source(self.options.wavfile, False)
  File "/usr/local/lib/python3/dist-packages/gnuradio/blocks/blocks_swig1.py", line 404, in make
    return _blocks_swig1.wavfile_source_make(filename, repeat)
RuntimeError: can't open file

What to do?
Thanks 
73's de Enzo IK8OZV

daniestevez, 2020-07-12:
As you may see, there is no recording from Nayif-1 in satellite-recordings. I'm always open to other people submitting suitable sample recordings to be included.

ik8ozv, 2020-07-12:
Dani,
what you mean that you need a recording piece?
How to record? is there a command so i can try to make a record
73's de Enzo IK8OZV

daniestevez, 2020-07-19:
Recordings in satellite recordings should be short 48ksps mono int16 WAV files containing one or a few valid packets. For BPSK 1k2 signals the signal should be centered at 1.5kHz inside the passband. You can make the recording however you like.

daniestevez, 2020-08-09:
I'm closing this, since there haven't been any additional questions in several weeks. Please feel free to reopen for more questions.

closed 2020-08-09 by daniestevez