daniestevez/gr-satellites #130: v3.1 'Telemetry Submit' block fails to SatNogs

K4KDR, 2020-07-13:
My first attempt to integrate the new 'Telemetry Submit' block results in the following error in the GRC monitor window:

`Error while submitting telemetry: HTTP Error 400: Bad Request`

The contents of my **~/.gr_satellites/config.ini** include:

```
[Groundstation]
callsign = K4KDR
latitude = 37.78
longitude = -77.61
submit_tlm = yes
```

It occurred to me that the 'Telemetry Submit' block also required the **NORAD ID** of the satellite, but since I am not using the 'Satellite Decoder' block, there is no reference to the satellite .YAML file (which includes the NORAD ID).

I tried passing several variations of "**--norad 45857**" in the 'Command line options' section of the 'Telemetry Submit' block without success.  

In contrast, the old 'Telemetry Forwarder' block works perfectly in the same flowgraph (as long as the required 'Parameter' values are supplied that it requires).

Thanks in advance for direction on how to properly use the new 'Telemetry Submit' block!

-Scott,  K4KDR

daniestevez, 2020-07-13:
Hi Scott,

This is in fact a bug. There should be a field to enter the NORAD ID that appears whenever "SatNOGS DB" is selected as server (and also for the BME ground station).  However, I see that this field doesn't appear for me either, so the NORAD ID can't be entered, which explains the failure.

I think this was once working correctly, so something has broken. I'll have to look into it in more detail.

daniestevez, 2020-07-13:
I've just pushed a commit to master that fixes this bug. Please check if it also works for you and if not reopen the issue.

closed 2020-07-13 by daniestevez

K4KDR, 2020-07-13:
Works perfectly; upload to SatNogs confirmed w/ recorded data!  Thank you so much!

I realize it's not a priority item, but since it is so very common to decode telemetry on the replay of recorded data, it would be ideal if either the '**Telemetry Submit**' block or the '**Print timestamp**' block incorporated a method to manually enter the time stamp of choice.

daniestevez, 2020-07-13:
Hi Scott,

Thanks for testing. What you say about the timestamps is already on the TODO list (see #92). However, this is not so easy to do. My idea is that the use should be able to enter the starting time of the recording, and then everything should magically work (even if the recording is played back faster than real time). However, I need GNU Radio to generate an propagate appropriate timestamps through the whole flowgraph.

Additionally, for the case of BY70-2, the timestamping is done by the proxy and by the server, so there is nothing I can do in gr-satellites to improve the playback of recordings.

K4KDR, 2020-07-13:
Very good - thank you so much for the update & info.

commit c0f9c0d2 (2020-07-13): Fix bug in Telemetry Submit that hid the NORAD ID field

It turns out that server == '"SatNOGS"' rather than 'SatNOGS',
so the comparison returned false and the NORAD ID field was always
hidden.

This fixes #130