daniestevez/gr-satellites #132: How to add Satellites

ik8ozv, 2020-07-14:
Hello everyone,
please i would like to add some satellites that are not anymore in the gr-satellites.
i mean if perhaps i give the command:

gr_satellites GO-32 --udp --samp_rate 48e3

i get the Error:

satellite not found

i tried to have a look in:

gr-satellites/python/satyaml

and actually there is no GO-32.yml.how to do if i want to try to receive the satellite and decode it.
I have used the GO-32 for the example.
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-14:
Adding a satellite is as easy as writing a new [SatYAML](https://gr-satellites.readthedocs.io/en/latest/satyaml.html) file. The SatYAML file of a similar satellite might be used as a basis and modified. If the satellite is not available in gr-satellites, please contribute the new SatYAML file in a pull request.

In the case of GO-32 I see [here](https://www.pe0sat.vgnet.nl/satellite/amateur-radio-satellites/go-32/) that it uses 9k6 FSK AX.25, so a SatYAML file should be easy to add by copying and modifying the SatYAML file of any other satellite that uses that mode.

ik8ozv, 2020-07-14:
Thanks Dani,
it is what i have done but i get always that the satellite is not find.
What am i doing wrong please?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-14:
The new SatYAML file either needs to be bundled and installed with the rest of the SatYAML files in gr-satellites or (the simplest approach), the path to the SatYAML file can be specified directly in `gr_satellites` and/or the Satellite decoder block (see [this](https://gr-satellites.readthedocs.io/en/latest/command_line.html#specifying-the-satellite)).

ik8ozv, 2020-07-14:
Please what i need to do?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-19:
A SatYAML file for GO-32 has been included in f730eb9558bbd8222003c316dfc32baf19945c94, and since I have also described how to create and use SatYAML files in general, I'm closing this.

closed 2020-07-19 by daniestevez

ik8ozv, 2020-07-28:
Dani please will you tell me the correct commad to bundle and install the new SatYAML file with the rest of the SatYAML files in gr-satellites.?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-28:
Just reinstall a newer version of gr-satellites on top of the old one.