daniestevez/gr-satellites #134: Question

ShayanMajumder, 2020-07-16:
Sir can you suggest a protocol in your gr-satellite project that can support bit rate of around 50 kilo bytes/sec and frequency in S band (2.4GHz specifically).

daniestevez, 2020-07-16:
What do you mean? Is this used by any particular satellite? Do you have a particular protocol in mind?

ShayanMajumder, 2020-07-17:
This is for my university satellite project. Initially we were planning to use 434Mhz frequency and AX.25 protocol.

However since we have have a imaging payload at this frequency we get really low downlink rates. So we are planning to switch to 2.4 Ghz for our downlink(Which is a legal amateur spectral band in India) . However we find that AX.25 gives maximum rate of around 9600 bits/sec. So we won’t use it in 2.4Ghz case.

Please suggest some protocols that we might be able to use effectively in S band and also which provides less bit error rate and high data rates.

We were thinking of NGHam however after doing some research we found though it outperforms AX.25 however it can’t provide the data rate we require. 

Thanks

daniestevez, 2020-07-17:
First of all, since this is for a university satellite project, I invite you to read the [note to satellite teams](https://github.com/daniestevez/gr-satellites/blob/master/satellite_teams.md) and maybe write a small presentation in the [issues](https://github.com/daniestevez/gr-satellites/issues/), as the [BeliefSat-1 team has done](https://github.com/daniestevez/gr-satellites/issues/112), to discuss these ideas about the S-band downlink further.

Your claim that AX.25 gives a maximum rate of 9600 bits/sec is simply not correct. The protocol (AX.25) and the data rate are two completely different things, and you can have AX.25 at really huge data rates if you want to. The problem with AX.25 is that it doesn't have any FEC. Therefore, it shouldn't be used for any space application (or any other serious endeavour), since just a single bit error will render the frame useless.

My recommendation is that you use standard CCSDS protocols and recommendations wherever possible. So you may begin by studying the TM synchronization & channel coding [blue](https://public.ccsds.org/Pubs/131x0b3e1.pdf) and [green](https://public.ccsds.org/Pubs/130x1g3.pdf) books. Something like QPSK using a concatenated code would be a reasonable starting point to think about possible solutions.

You might want to take a look at the [S-band receiver](https://destevez.net/2018/07/k2sat-s-band-image-receiver/) I did for K2SAT. This is available in gr-satellites. The protocol that the K2SAT S-band transmitter used is not perfect, and I would have changed it in several aspects, but this is also a good example.

Finally, maybe before thinking what protocol you are going to use, you should think about the following things: what data rate you need/want, how much margin do you have in your link budget, what hardware you'll use in the spacecraft. All these conditions affect in one way or another the choice of the communications system.

ShayanMajumder, 2020-07-18:
Sir, first of all I would like to thank you for your prompt reply.
Regarding the AX.25 part we read it in some paper that it can’t handle data rates more than 9600 bits/sec same is written in its Wikipedia page under its limitations.
We will have a full team meet and get in touch to you in a week or so.
Also is the K2SAT working in the S band? Since one of the reasons of failing of Aalto-1 in transmitting in 2.4 GHz was due to poor pointing efficiency. For your reference our satellite is 3U-CubeSat carrying a hyper-spectral camera as payload.

daniestevez, 2020-07-18:
Wikipedia mentions

> At the speeds commonly used to transmit packet radio data (rarely higher than 9,600 bit/s, and typically 1,200 bit/s)

This is true. Packet radio is often used at baudrates between 1k2 and 9k6. But this doesn't mean anything about AX.25 but rather about one of its applications.

I'm not sure if the K2SAT team was ever able to use the S-band transmitter successfully. The satellite had some problems since the beginning of the mission.

closed 2020-07-23 by ShayanMajumder