daniestevez/gr-satellites #137: Unknown frame type. Not processing frame.

ik8ozv, 2020-07-19:
Hello today,
i have tried to decode JY1SAT ( JO-97 ) giving the following command:

gr_satellites JY1-Sat --udp --samp_rate 48e3

but i have noticed that on each frame i get the following error:

Unknown frame type. Not processing frame.

What am i doing wrong?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-19:
You've done nothing wrong. These are most likely frames which the telemetry parser can't process because I don't have the format details for them. You can view them in hex by using `--hexdump`. Perhaps they are SSDV frames which are decodable with the JY1SAT SSDV decoder, which is not functional in v3.2.0, but it is working on master.

More information [here](https://community.libre.space/t/decoding-nayif-1-with-satnogs-ogg-files-in-gr-satellites/6456/9?u=ea4gpz).

daniestevez, 2020-08-09:
I'm closing this, since there were no additional questions in several weeks. Please feel free to reopen for further questions.

closed 2020-08-09 by daniestevez