daniestevez/gr-satellites #138: --version displays same info as -h

K4KDR, 2020-07-19:
With the most recent MASTER commit to-date (eef6199), the following command:

`gr_satellites --version`

displays the same command help as:

`gr_satellites -h`

... with no mention of the installed version number.

Thanks!

daniestevez, 2020-07-19:
Hi Scott,

For technical reasons, `gr_satellites` is always intended to be run as `gr_satellites SATELLITE [options]`, so if you run `gr_satellites --version` it is actually complaining that you need to specify a satellite. So the correct way is to specify some satellite. For instance `gr_satellites AO-73 --version` (in the same way, note that `gr_satellites --help` and `gr_satellites AO-73 --help` don't show the same thing).

But maybe this is completely non-obvious, so I might think of a workaround.

K4KDR, 2020-07-19:
Thank you!!  My mistake.   But yes, I suppose **--version** is somewhat of a standard for many/most linux commands

commit 2f36b324 (2020-08-19): Show version if gr_satellites --version is run

Due to the way that gr_satellites uses argparse, previously it
was necessary to run "gr_satellites SAT --version" (where SAT
is the name of a valid satellite) in order to show the version.
This was a bit confusing.

Now "gr_satellites --version" is also added as an special case
in order to achieve the expect result.

This closes #138
closed 2020-08-19 by daniestevez