daniestevez/gr-satellites #139: Aistechsat-2 error

ik8ozv, 2020-07-20:
Hello,
Today i have tried to receive and in the meantime to decode the AISTECHSAT-2.
I give the command:

gr_satellites AISTECHSAT-2 --udp --samp_rate 48e3

but on each packet i receive i get the following error:

Could not parse telemetry beacon Error in path (parsing) -> space_packet_header
stream read less than specified amount, expected 6, found 0

Please have i done something incorrect?
How to solve?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-20:
I think that this might just be false decodes, which the telemetry parser can't parse because they are too short. So the error message you get is to be expected.

It might also be a regression bug in the telemetry parser. I don't have a recording for AISTECHSAT-2 or LUME-1 (they use the same telemetry format) to test it with. If you can record frames with `--hexdump` then these can be used to test the decoder.

daniestevez, 2020-08-09:
I'm closing this, since the @ik8ozv didn't report any additional information in several weeks. Please feel free to reopen with further information.

closed 2020-08-09 by daniestevez