daniestevez/gr-satellites #140: Installation on WSL Debian

doctorvenables, 2020-07-22:
After pulling the newest version and building from source, on my version of WSL, installation puts libgnuradio-satellites.so etc in a different directory: for whatever reason it puts it into /usr/local/lib/x86_64-linux-gnu

Adding this directory to PYTHONPATH and sudo ldconfig seems to fix it and test.sh works

The qa tests still fail but I'll look into that later. I solved it for 3.2.0 but forgot what I did.

daniestevez, 2020-07-22:
That's interesting. I don't have WSL, so I can't test this, but please keep me updated and let me know if there is anything I can do to improve compatibility.

Regarding the problem with the tests, these should run with the `.so` file and Python module that are in the `build/`directory, even before `make install` is run. Getting this to work is a bit tricky, so I guess it could give problems in some platforms. However, your problem with the tests might be #116. Note that the fix has already been merged to GNU Radio master, but most likely it isn't available yet in the version of GNU Radio you're using.

doctorvenables, 2020-07-22:
Curious - qa tests work now and I'm not sure I did anything. I'll close this for now and perhaps re-open if it happens ago consistently in future.

closed 2020-07-22 by doctorvenables