daniestevez/gr-satellites #142: Error while loading /usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.ym

alphafox02, 2020-07-25:
Not sure if this is harmless, but here's the output loading gnuradio-companion on 20.04 with gnuradio 3.8 from repository with today's pull of gr-satellites. 

gnuradio-companion 
ERROR:gnuradio.grc.core.platform:Error while loading /usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/platform.py", line 169, in build_library
    data = cache.get_or_load(file_path)
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/cache.py", line 66, in get_or_load
    data = yaml.safe_load(fp)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 115, in check_token
    while self.need_more_tokens():
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 152, in need_more_tokens
    self.stale_possible_simple_keys()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 291, in stale_possible_simple_keys
    raise ScannerError("while scanning a simple key", key.mark,
yaml.scanner.ScannerError: while scanning a simple key
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 41, column 1
could not find expected ':'
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 42, column 29
ERROR:gnuradio.grc.core.platform:while scanning a simple key
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 41, column 1
could not find expected ':'
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 42, column 29
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/platform.py", line 169, in build_library
    data = cache.get_or_load(file_path)
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/cache.py", line 66, in get_or_load
    data = yaml.safe_load(fp)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 115, in check_token
    while self.need_more_tokens():
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 152, in need_more_tokens
    self.stale_possible_simple_keys()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 291, in stale_possible_simple_keys
    raise ScannerError("while scanning a simple key", key.mark,
yaml.scanner.ScannerError: while scanning a simple key
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 41, column 1
could not find expected ':'
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 42, column 29
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
        /usr/share/gnuradio/grc/blocks
        /usr/local/share/gnuradio/grc/blocks
>>> Check: /usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml
>>> FlowGraph Error: while scanning a simple key
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 41, column 1
could not find expected ':'
  in "/usr/local/share/gnuradio/grc/blocks/satellites_lucky7_deframer.block.yml", line 42, column 29

alphafox02, 2020-07-25:
Easy fix, there's a random letter d hanging out on line 41

daniestevez, 2020-07-26:
Thanks for noticing this. I don't know why it wasn't giving me problems. I've seen that `cmake` doesn't reinstall `grc` files sometimes, so maybe that's why it worked on my system.

commit 595cecbf (2020-07-26): Remove stray line in Lucky7 deframer GRC block

This fixes #142
closed 2020-07-26 by daniestevez