daniestevez/gr-satellites #143: Solution for the issue of ModuleNotFound error

rnbokade, 2020-07-26:
The export Python Path solution you mentioned in the docs works well for the terminal based calls, but if you open gnu radio companion and then use the gr-satellite blocks the problem still persisted for me at least. Turns out there are 2 different locations
1. "/usr/local/lib/python3/dist-packages" - This is where the package gets installed generally
2. "/usr/lib/python3/dist-packages"- This is where the packages need to be to work properly even with the other applications.
so just 
`sudo cp -i -r /usr/local/lib/python3/dist-packages/satellites usr/lib/python3/dist-packages`
This fixes the problem for me. If it does for others too, @daniestevez you may wanna add this to documentation.

daniestevez, 2020-07-26:
Hi Rohit,

While the workaround you suggest certainly works, I don't think it's the best solution. It would be better to export the correct `PYTHONPATH` in a way such that graphical applications such as `gnuradio-companion` can see it, such as in `xinitrc`, as described by the [Arch Linux documentation](https://wiki.archlinux.org/index.php/Environment_variables#Graphical_environment).

If you want gr-satellites installed into `/usr/` instead of  `/usr/local/` so that you don't need to touch the `PYTHONPATH`, then you can tell `cmake` to do so by using [CMAKE_INSTALL_PREFIX](https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX.html).

daniestevez, 2020-08-09:
@rnbokade did you find these solutions reasonable? I'm trying to see if this issue can be closed.

rnbokade, 2020-08-09:
Yes those were reasonable. I think you should mention theses in docs, before closing. Or as a side note somewhere, so that it helps somebody with similar issue.

commit d726c008 (2020-08-19): Add documentation about permanently setting environment variables

This explains .bash_profile, .bashrc, xinitrc, xprofile, etc,
linking the help from Arch Linux.

This closes #143
closed 2020-08-19 by daniestevez