daniestevez/gr-satellites #146: Question

luisguarnizo1234, 2020-07-28:
hola Daniel, 

Antes que nada quería agradecerte por compartir este contenido para que otras personas podamos aprender de tus experiencias.

Tengo una pregunta sobre como determinar el formato de beacon para un satélite determinado. estuve leyendo tu publicación "decoding-gomx-1-telemetry" donde explicabas como había sido el proceso para decodificar las tramas del gomx-1, aquí hacias referencia a que Mike DK3WN te habia proporcionado el formato de baliza y recomendabas revisar gomx-1-beacon.c. Quisiera saber si me podrías orientar sobre como encontrar el formato de los beacons para el satélite FACSAT-1, ya que probé con el gr-satellite y obtuve el encabezado CSP pero no pude observar el payload.

Te agradezco por tu amable atención, ¡Saludos!

daniestevez, 2020-07-29:
Hola,

Revisando mis emails, veo que tengo el siguiente documento donde se describe el formato de la baliza. 

[FACSat 1 Beacon headers.pdf](https://github.com/daniestevez/gr-satellites/files/4997455/FACSat.1.Beacon.headers.pdf)

Al parecer en su día encontramos alguna errata en el documento. También observamos que el formato era muy similar a LUME-1 y AISTECH-2. Básicamente es el formato de telemetría que ofrecen los módulos de GOMspace.

Nunca llegué a implementar esta descripción de formato en gr-satellites, no recuerdo exactamente por qué. Si estás interesado en ello, puedes comprobar hasta qué punto es correcto el documento, comparando con tramas transmitidas por FACSAT-1 y quizá con la descripción de otros satélites e implementar la descripción del formato en gr-satellites.

luisguarnizo1234, 2020-08-25:
hola Dani,

he creado un archivo para definir la telemetría de facsat-1, hice una copia del archivo lime.py que esta en la carpeta de telemetry y le cambié el nombre por facsat.py. al momento de intentar decodificar la grabación facsat.wav que tienes en la carpeta de recordings me arroja esta respuesta de la imagen, pero no se donde debo el modulo satellites.telemetry.

intente en la carpeta componentes, datasinks editar el archivo CmakeList y agregar a facsat.py pero no dio resultado

[image]

daniestevez, 2020-08-25:
Hola Luis,

Para hacer lo que pretendes debes modificar [esta línea](https://github.com/daniestevez/gr-satellites/blob/master/python/telemetry/lume.py#L167) en `facsat.py` y escribir `facsat` en lugar de `lume`. Además, debes añadir
```
from .facsat import facsat
```
en [`telemetry/__init__.py`](https://github.com/daniestevez/gr-satellites/blob/master/python/telemetry/__init__.py) así como añadir `facsat.py` en [`telemetry/CMakeLists.txt`](https://github.com/daniestevez/gr-satellites/blob/master/python/telemetry/CMakeLists.txt). Estos archivos están ordenados alfabéticamente, así que es bueno respetar el orden.

luisguarnizo1234, 2020-08-25:
muchas gracias Dani, 

ayer hice una prueba con el decodificador de gomx-1 y me funcinó bien.Sin embargo soy cociente que no tienen el mismo formato de telemetría. Intentare agregar el decoficador con las instrucciones que me diste y te estaré comentando como me va.

Muchas gracias por tu apoyo Dani.

estos fueron los resultados para facsat-1 con el decodificador de gomx-1
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

luisguarnizo1234, 2020-08-25:
muchas gracias Dani,

ayer hice una prueba con el decodificador de gomx-1 y me funcionó bien. Sin embargo soy cociente que no tienen el mismo formato de telemetría. Intentare agregar el decoficador con las instrucciones que me diste y te estaré comentando como me va.

Muchas gracias por tu apoyo Dani.

luisguarnizo1234, 2020-08-25:
hola Dani,

como te mencioné, he utilizado el decodificador del gomx-1 para la grabación facsat.wav y obtuve el resultado de la primera imagen. Ya modifique los archivos según tus instrucciones, pero en lugar de usar el lume.py usé el gomx_1.py y funciona perfecto. Sin embargo, cuando quiero modificar la sección de "com" en la estructura presenta un error como te muestro en la segunda imagen.

como puedes ver, el beacon de 1981-09-12 17:30:08 ya no muestra toda la estructura de la telemetria (obc, com, eps). solo intenté modificar el "com"
[image]
[image]

 porque según el manual del radio AX100 estoy seguro que el formato es correcto.

daniestevez, 2020-08-26:
Hola Luis,

Perdón, pero no entiendo bien tu último mensaje. No sé qué quieres transmitir con esas dos imagenes o qué no funciona o cuál es tu pregunta. Por favor intenta ser más concreto.

luisguarnizo1234, 2020-08-26:
hola Dani, 

Lo que quiero decir es que cuando modifico la estructura de la telemetría en el archivo gomx-1.py y ejecuto nuevamente gr-satellites para decodificar la grabación de facsat.wav, no se aplican los cambios que realice. En el resultado de la terminal simplemente ya no aparece información sobre el obc, com, la eps, solamente aparece la información del csp.

lo que yo quiero hacer es modificar la estructura de la telemetría del gomx-1.py con base en el documento que me enviaste en el que se muestra el formato de telemetria de facsat-1.

daniestevez, 2020-08-26:
Sigo sin entender:

* Dices que modificas algo
* Dices que después de modificar esto, *"simplemente ya no aparece información sobre el obc [...]"*, de modo que entiendo que tu modificación tiene algún efecto sobre el comportamiento de gr-satellites
* Sin embargo, dices *"no se aplican los cambios que realice"*. No entiendo cómo esto es compatible con el punto anterior. Dado que se según el punto anterior, sí cambia el comportamiento de gr-satellites, ¿es posible que el cambio que observas no sea el que esperabas?

De nuevo, pido por favor, ¿puedes intentar ser más claro? Quizás resultado esperado vs. resultado obtenido sea una buena manera de organizar la información en este caso (describiendo con todo lujo de detalles).

Adicionalmente, aunque no me comprometo a revisar código, creo que sería conveniente que compartas el código que estás usando. Si no, es complicado saber exactamente qué estás haciendo. Puedes usar una nueva rama en tu fork de gr-satellites para compartir el código en el que estás trabajando.

luisguarnizo1234, 2020-08-26:
hola dani, gracias por tu paciencia. Espero que esta vez consiga expresar adecuadamente mi problema.

gracias de antemano por toda tu ayuda.

RESULTADO ESPERADO:

el resultado que espero es que se modifique la sección del COM en el formato de la telemtría. Quiero que al ejecutar gr-satellites en la parte del COM se muestre el formato de telemetría para el COM del facsat-1 en lugar del formato de telemetría para el COM del gomx-1.

este es el código original del gomx_1.py, en la sección del COM

`COM = Struct(
    'byte_corr_tot' / Int16ub, # Bytes corrected by RS
    'rx' / Int16ub, # RX packets
    'rx_err' / Int16ub,
    'tx' / Int16ub,
    'last_temp' / Int16sb[2],
    'last_rssi' / Int16sb, # dBm
    'last_rferr' / Int16sb, # Hz
    'last_batt_volt' / LinearAdapter(100.0, Int16ub), # V
    'last_tx_current' / Int16ub, # mA
    'boot_count' / Int16ub
    )
`
yo cree una copia del archivo gomx_1.py y le llame facsat.py y modifiqué unicamente la sección del COM de la siguiente manera,  teniendo en cuenta el formato de telemetria que esta en el documento que me enviaste:

`COM = Struct(
    'temp_brd' / Int16sb, # Bytes corrected by RS
    'temp_pa' / Int16sb, # RX packets
    'last_rssi' / Int16sb,
    'last_rferr' / Int16sb,
    'tx_count' / Int32ub,
    'rx_count' / Int32ub,
    'tx_bytes' / Int32ub,
    'rx_bytes' / Int32ub,
    'active_conf' / Int8sb,
    'boot_count' / Int16ub,
    'boot_cause' / Int32ub,
    'last_contact' / Int32ub,
    'bgnd_rssi' / Int16sb,
    'tx_duty' / Int8ub,
    'tot_tx_count' / Int32ub,
    'tot_rx_count' / Int32ub,
    'tot_tx_bytes' / Int32ub,
    'tot_rx_bytes' / Int32ub
    )`

================================================================================
RESULTADO OBTENIDO:

1. antes de modificar el codigo obtengo este resultado para la grabación de facsat.wav, en el cual se consigue decodificar la información del COM, pero en el formato del gomx-1
===============================================
Container: 
    csp_header = Container: 
        priority = 2
        source = 4
        destination = 28
        destination_port = 30
        source_port = 0
        reserved = 0
        hmac = False
        xtea = False
        rdp = False
        crc = True
    beacon_time = 1981-09-12 17:38:08
    beacon_flags = 181
    beacon = Container: 
        obc = Container: 
            boot_count = 2873
            temp = ListContainer: 
                -194.25
                -2048.0
            panel_temp = ListContainer: 
                256.0
                0.0
                0.0
                0.0
                0.0
                0.0
        com = Container: 
            byte_corr_tot = 63
            rx = 32768
            rx_err = 0
            tx = 0
            last_temp = ListContainer: 
                0
                0
            last_rssi = 0
            last_rferr = 0
            last_batt_volt = 0.0
            last_tx_current = 0
            boot_count = 0
        eps = Container: 
            vboost = ListContainer: 
                0.0
                0.0
                0.0
            vbatt = 0.0
            curout = ListContainer: 
                0
                0
                0
                0
                0
                0
            curin = ListContainer: 
                0
                0
                0
            cursun = 0
            cursys = 0
            temp = ListContainer: 
                0
                0
                0
                0
                0
                0
            output = 0
            counter_boot = 0
            counter_wdt_i2c = 0
            counter_wdt_gnd = 0
            bootcause = 0
            latchup = ListContainer: 
                0
                0
                0
                0
                0
                0
            battmode = (enum) normal 0
        gatoss = Container: 
            average_fps_5min = 0
            average_fps_1min = 0
            average_fps_10sec = 0
            plane_count = 0
            frame_count = 0
            last_icao = 0x00000000
            last_timestamp = 1970-01-01 00:00:00
            last_lat = 0.0
            last_lon = 0.0
            last_altitude = 0
            crc_corrected = 0
            boot_count = 0
            boot_cause = 0
        hub = Container: 
            temp = 0
            boot_count = 0
            reset_cause = 0
            switch_status = 0
            burns = ListContainer: 
                0
                0
        adcs = Container: 
            tumblerate = ListContainer: 
                0.0
                0.0
                -7994821775982592.0
            tumblenorm = ListContainer: 
                -9.086521978316027e+33
                0.0
            magnetometer = ListContainer: 
                0.0
                0.0
                0.0
            status = 0
            torquerduty = ListContainer: 
                0.0
                0.0
                0.0
            ads_state = 0
            acs_state = 0
            sunsensor = ListContainer: 
                0
                218
                43
                204
                105
                29
                234
                48

===============================================

2. una vez realizo las modificaciones de la sección del COM en el archivo facsat.py, ejecuto nuevamente el gr-satellites y el resultado que obtengo es que solamente se decodifica la información referente al CSP, pero no puedo decodificar la información del COM. 

Container: 
    csp_header = Container: 
        priority = 2
        source = 4
        destination = 28
        destination_port = 30
        source_port = 0
        reserved = 0
        hmac = False
        xtea = False
        rdp = False
        crc = True
    beacon_time = 1981-09-12 17:38:08
    beacon_flags = 181
    beacon = None

daniestevez, 2020-08-26:
> yo cree una copia del archivo gomx_1.py y le llame facsat.py y modifiqué unicamente la sección del COM de la siguiente manera, teniendo en cuenta el formato de telemetria que esta en el documento que me enviaste:

Tal como lo has descrito, esto no es correcto. Tienes que tomar en consideración mi [comentario](https://github.com/daniestevez/gr-satellites/issues/146#issuecomment-680119109) (aplicado a `gomx_1` en lugar de `lume`). No estoy seguro de si lo has hecho.

Adicionalmente, revisando el parser de GOMX-1, veo que la [línea para `beacon_a`](https://github.com/daniestevez/gr-satellites/blob/master/python/telemetry/gomx_1.py#L104) está marcada con `Optional`. Quizá este sea tu problema. Si el tamaño total del `Struct` que has modificado excede del tamaño de la trama que estás procesando, el campo `beacon_a` no se procesará y obtendrás `beacon = None`, justo como te ocurre.

luisguarnizo1234, 2020-08-26:
hola dani, 

Si seguí tus indicaciones (en telemetry/__init__.py así como añadir facsat.py en telemetry/CMakeLists.txt. Estos archivos están ordenados alfabéticamente, así que es bueno respetar el orden.). Esta parte funciona correctamente, ya lo comprobé. 

con respecto al tamaño de la trama beacon_a, como podría modificar el tamaño para que acepte la trama del facsat1. En cual archivo deberia hacer esa modificación.

daniestevez, 2020-08-26:
Voy a decirlo de otra forma: al haber modificado la definición del `Struct` para COM, la trama que tienes definida a través de structs de construct es demasiado larga.

Como dije, creemos que el documento que estás usando tiene erratas, así que no está garantizado que implementándolo al pié de la letra vaya a funcionar. Una comprobación básica: ¿has comprobado manualmente si el tamaño de las tramas, según las describe el documento cuadra con el tamaño de las tramas presentes en la grabación que estás usando? Si no cuadra, el documento no es correcto.

luisguarnizo1234, 2020-08-26:
gracias Dani, entiendo lo que me dices. 

No he hecho la comprobación que me dices, no sabría como realizar esta comprobación. ¿Tal vez en alguno de tus blogs tratas este tema? o me podrías indicar de que forma se podría realizar.

muchas gracias por tu apoyo Dani

daniestevez, 2020-08-26:
El tamaño de las tramas que hay en la grabación de FACSAT-1 supongo que ya lo conoces. Si no, puedes usar la opción `--hexdump` de gr-satellites.

El tamaño de las tramas según las describe el documento lo obtienes simplemente sumando (a mano) las longitudes de los campos allí descritos.

luisguarnizo1234, 2020-08-28:
hola dani, 

Realice la prueba que me dijiste para verificar el tamaño de las tramas pero según veo el tamaño varia, ya que al ejecutar el comando gr_satellites FACSAT-1 --wavfile satellite-recordings/facsat_1.wav --sam_rate 48e3 --hexdump obtengo diferentes tamaños para las tramas. Por ejemplo dentro de la misma grabación obtengo los siguiente pdu_length.

pdu_length= 51
pdu_length= 119
pdu_length= 78
pdu_length= 185
pdu_length= 113
pdu_length= 176
pdu_length= 217
pdu_length= 180

daniestevez, 2020-09-28:
I'm closing this, since it is already open for a month without any further feedback. If you resume work on this, please feel free to re-open.

closed 2020-09-28 by daniestevez