daniestevez/gr-satellites #147: BY70-2

ik8ozv, 2020-08-01:
Hello all.
please anybody can tell me how to decode the BY70-2 Satellite?
Where can i find the needed stuff?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-01:
Support for decoding BY02 is available since v3.1.0. There are some improvements related to BY02 in in further versions, which you may see in the [Changelog](https://github.com/daniestevez/gr-satellites/blob/master/CHANGELOG.md). It is decoded just like any other satellite supported by gr-satellites. More information in [this post](https://destevez.net/2020/07/by02-telemetry-beacon/).

ik8ozv, 2020-08-09:
Hello daniel,
i am noticing a strange behavior on the BY70-2 reception. i ran the command:

gr_satellites BY02 --udp --samp_rate 48e3 --hexdump --kiss_out /home/enzo/Scrivania/BY70-2.kss

But i notice to get the data in the terminal window when there is no signal.
I mean i have just tuned the GQRX where there was non sinal and exactly at 435.000 and i could see
the data still incoming in the terminal window.
Please it is a bug or am i doing something wrong?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-09:
This is expected. See [this post](https://destevez.net/2020/07/by02-telemetry-beacon/), especifically

> An important remark about --syncword_threshold is in order. The default syncword detection threshold allows up to 4 bit errors in the 32 bit syncword. Since there is no checksum or Reed-Solomon to throw invalid frames, this will produce false decodes and partially corrupted frames. This might be good when one wants to copy the Codec2 data in difficult conditions, but for decoding good telemetry frames, a lower threshold is recommended to prevent false decodes.

ik8ozv, 2020-08-09:
Please can you be so kind to tell me the correct command to give in the terminal?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-17:
Duplicate of #152

closed 2020-08-17 by daniestevez