daniestevez/gr-satellites #150: v3.3 error:  "Could not find port: frame in:"

K4KDR, 2020-08-11:
After upgrading gr-satellites to v.3.3.0, my existing flowgraph for BY70-2 fails with a series of errors that I am not quite able to understand.

Please note also that even the '--version' command results in a core dump.

**Version info:**

```
k4kdr@k4kdr-u1804:~/gr-satellites$ git branch
* master
k4kdr@k4kdr-u1804:~/gr-satellites$ 

k4kdr@k4kdr-u1804:~/gr-satellites$ gr_satellites AO-73 --version
gr_satellites v3.3.0-git
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
double free or corruption (fasttop)
Aborted (core dumped)
k4kdr@k4kdr-u1804:~/gr-satellites$ 
```

**GNU Radio Companion output:**

```
Generating: '/home/k4kdr/grc/modified/BY70_2_38.py'

Executing: /usr/bin/python3.6 -u /home/k4kdr/grc/modified/BY70_2_38.py

Could not find port: frame in:
system

Traceback (most recent call last):
  File "/home/k4kdr/grc/modified/BY70_2_38.py", line 243, in <module>
    main()
  File "/home/k4kdr/grc/modified/BY70_2_38.py", line 221, in main
    tb = top_block_cls()
  File "/home/k4kdr/grc/modified/BY70_2_38.py", line 84, in __init__
    self.satellites_lilacsat_1_deframer_0 = satellites.components.deframers.lilacsat_1_deframer(syncword_threshold = 0, options="")
  File "/usr/local/lib/python3.6/dist-packages/satellites/components/deframers/lilacsat_1_deframer.py", line 54, in __init__
    self.msg_connect((self.demux0, 'frame'), (self, 'out'))
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 3665, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()
double free or corruption (!prev)

>>> Done (return code -6)
```

The flowgraph in question that I have been using successfully prior to this upgrade appears as follows (telemetry forwarder blocks temporarily disabled for testing):

https://www.dropbox.com/s/tz06bxhr3me1f3v/k4kdr-by70-2-v5.png?raw=1

daniestevez, 2020-08-11:
Hi Scott,

The new version seems to work for me, so I wonder what is happening. A few preliminary comments.

You are running from the master branch rather than the v3.3.0 stable version. Since v3.3.0 was released just today, the two are pretty much the same, but be aware that with time new modifications will make their way into master and be eventually released as v3.4.0. When the master branch, you can report the exact commit that you're using by running `git rev-parse HEAD` inside the `gr-satellites` folder. With that information I can be sure that I'm testing exactly the same code as you.

I'm testing 0d2e7dda5984971708cfd290778bbc2357b7add1, which I think is the same you're running.

How did you upgrade? If you had files from a previous build in the `build/` folder, maybe cmake was not smart enough to figure out what needs to be rebuilt. You can try deleting everything inside `build/` and then running 
```
cmake .. && make && sudo make install && sudo ldconfig
```

If you find problems such as these, run the tests with `make test` inside the `build/` folder, and if some of them fail, run `ctest -V` to get a more detailed output. This can help catch several problems.

For me, `gr-satellites AO-73 --version` is running correctly, as well as
```
gr_satellites BY02 --wavfile satellite-recordings/by02.wav --samp_rate 48e3
```

You can attach the .grc flowgraph you're using in one of these Github messages, and so I could test it very easily.

By now, I think that what's happening is so strange that it is most likely a problem from the upgrade (leftover files from the previous version somewhere). If not, then we'll need to investigate this in more detail.

K4KDR, 2020-08-11:
Thank you so much for the reply!

Here is the build info:

```
k4kdr@k4kdr-u1804:~/gr-satellites$ git rev-parse HEAD
0d2e7dda5984971708cfd290778bbc2357b7add1
```

... and the upgrade was performed as follows:

(in ~/gr-satellites)

```
git pull -v
rm -rf build
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

The flowgraph in question is attached - many thanks!!

[by70-2-codec2-3.8-v5-k4kdr.grc--remove.txt](https://github.com/daniestevez/gr-satellites/files/5059161/by70-2-codec2-3.8-v5-k4kdr.grc--remove.txt)

K4KDR, 2020-08-11:
The 'ctest -V' command produces a similar error to my BY70-2 flowgraph:

```
test 4
    Start 4: qa_ao40_fec_deframer

4: Test command: /bin/sh "/home/k4kdr/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
4: Test timeout computed to be: 9.99988e+06
4: Could not find port: out in:
4: system
4: 
4: E
4: ======================================================================
4: ERROR: test_ao40_fec_deframer (__main__.qa_ao40_fec_deframer)
4: Loads symbols from an AO-73 packet and checks if ao40_fec_deframer produces the expected output and intermediate data (stored in a reference files)
4: ----------------------------------------------------------------------
4: Traceback (most recent call last):
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/qa_ao40_fec_deframer.py", line 89, in test_ao40_fec_deframer
4:     deframer = ao40_fec_deframer()
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/ao40_fec_deframer.py", line 53, in __init__
4:     self.msg_connect((self.deframer, 'out'), (self.deinterleaver, 'in'))
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
4:     func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
4:     self.primitive_msg_connect(*args)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 3665, in primitive_msg_connect
4:     return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
4: RuntimeError: invalid msg port in connect() or disconnect()
4: 
4: ----------------------------------------------------------------------
4: Ran 1 test in 0.005s
4: 
4: FAILED (errors=1)
4/7 Test #4: qa_ao40_fec_deframer .............***Failed    2.13 sec
```

daniestevez, 2020-08-11:
I confirm that your `.grc` file runs perfectly in my system.

Let's try to remove all the gr-satellites installation from your system and install again. You can do something like this
```
cd gr-satellites/build
sudo make uninstall
sudo rm -rf /usr/local/share/gnuradio/grc/blocks/satellites*
sudo rm -rf /usr/lib/python3.7/site-packages/satellites/
sudo rm -rf /usr/local/include/satellites/
sudo rm -rf /usr/local/lib/cmake/satellites/
sudo rm -rf /usr/local/lib64/libgnuradio-satellites*
```
Note the paths. They might be slightly different in your system. In particular, the python3.7 folder might be in `/usr/local/lib` or be another version of python3.

After this, please try to install again, first removing everything inside the build folder.

K4KDR, 2020-08-11:
With a 'sudo make uninstall' & all those references to /satellites deleted, a fresh compile in an empty /build folder resulted in the same failure on test #4 

```
    Start 4: qa_ao40_fec_deframer

4: Test command: /bin/sh "/home/k4kdr/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
4: Test timeout computed to be: 9.99988e+06
4: Could not find port: out in:
4: system
4: 
4: E
4: ======================================================================
4: ERROR: test_ao40_fec_deframer (__main__.qa_ao40_fec_deframer)
4: Loads symbols from an AO-73 packet and checks if ao40_fec_deframer produces the expected output and intermediate data (stored in a reference files)
4: ----------------------------------------------------------------------
4: Traceback (most recent call last):
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/qa_ao40_fec_deframer.py", line 89, in test_ao40_fec_deframer
4:     deframer = ao40_fec_deframer()
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/ao40_fec_deframer.py", line 53, in __init__
4:     self.msg_connect((self.deframer, 'out'), (self.deinterleaver, 'in'))
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
4:     func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
4:     self.primitive_msg_connect(*args)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 3665, in primitive_msg_connect
4:     return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
4: RuntimeError: invalid msg port in connect() or disconnect()
4: 
4: ----------------------------------------------------------------------
4: Ran 1 test in 0.034s
4: 
4: FAILED (errors=1)
4/7 Test #4: qa_ao40_fec_deframer .............***Failed    2.14 sec
```

K4KDR, 2020-08-12:
Tried numerous previous gr-satellites releases with the same error; so apparently not an issue specific to v3.3.0.

Performed the following:

```
sudo apt-get update
sudo apt-get upgrade
```

REBOOT

(system is Ubuntu 18.04)

... and attempted the upgrade to gnuradio v3.8.1  (which has previously failed to compile)

gnuradio v3.8.1 compiled & installed successfully

```
k4kdr@k4kdr-u1804:~$ gnuradio-config-info -v
3.8.1.0
k4kdr@k4kdr-u1804:~$ 
```

... and proceeded to compile a fresh copy of gr-satellites.

cmake & make completed without errors, but the result is the same for "ctest -V":

```
test 4
    Start 4: qa_ao40_fec_deframer

4: Test command: /bin/sh "/home/k4kdr/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
4: Test timeout computed to be: 9.99988e+06
4: Could not find port: out in:
4: system
4: 
4: E
4: ======================================================================
4: ERROR: test_ao40_fec_deframer (__main__.qa_ao40_fec_deframer)
4: Loads symbols from an AO-73 packet and checks if ao40_fec_deframer produces the expected output and intermediate data (stored in a reference files)
4: ----------------------------------------------------------------------
4: Traceback (most recent call last):
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/qa_ao40_fec_deframer.py", line 89, in test_ao40_fec_deframer
4:     deframer = ao40_fec_deframer()
4:   File "/home/k4kdr/gr-satellites/python/components/deframers/ao40_fec_deframer.py", line 53, in __init__
4:     self.msg_connect((self.deframer, 'out'), (self.deinterleaver, 'in'))
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
4:     func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
4:     self.primitive_msg_connect(*args)
4:   File "/usr/local/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 3665, in primitive_msg_connect
4:     return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
4: RuntimeError: invalid msg port in connect() or disconnect()
4: 
4: ----------------------------------------------------------------------
4: Ran 1 test in 0.005s
4: 
4: FAILED (errors=1)
4/7 Test #4: qa_ao40_fec_deframer .............***Failed    2.09 sec
```

K4KDR, 2020-08-12:
There seem to be many references to this error online that eventually all refer to:

https://github.com/gnuradio/gnuradio/pull/2543

... but unfortunately, I do not understand the discussion of port names & YML files.

K4KDR, 2020-08-12:
While I regret not resolving this the "right" way, in the interest of having a fully-functional system, I have restored the machine to a recent image that was working flawlessly with GNU Radio 3.8.0 & gr-satellites maint3.8 branch.   

Thank you for your assistance but I will suspend any further efforts on this error which was clearly specific to how the various apps evolved on my system to-date.

daniestevez, 2020-08-12:
Hi Scott, good that you were able to restore to something functional. Just a question: when you say the maint3.8 branch, is that an older version of maint3.8? Today maint3.8 has almost the same as master. If you still have the git repository from where you installed, you can do `git rev-parse HEAD` on it. This might help me identify what has been the change that broke things for you.

K4KDR, 2020-08-12:
I believe that I'm up-to-date in that branch as of the current date:

```
k4kdr@k4kdr-u1804:~$ cd gr-satellites

k4kdr@k4kdr-u1804:~/gr-satellites$ git branch
* maint-3.8

k4kdr@k4kdr-u1804:~/gr-satellites$ git rev-parse HEAD
d501b9c60f3260d8536d5a117f822154b2652b3a

k4kdr@k4kdr-u1804:~/gr-satellites$ gr_satellites AO-73 --version
gr_satellites v3.3.0
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
k4kdr@k4kdr-u1804:~/gr-satellites$ 
```
... if this had been a clean system, no doubt all would be fine.  But instead it has evolved through many versions of GNU Radio, python, and gr-satellites.

daniestevez, 2020-08-12:
Well, so it seems you're running the latest stable version from gr-satellites. That's basically identical to what we have now in master. The main difference is that master identifies as `v3.3.0-git`.

So it seems that rolling back your system has fixed whatever weird problem it had. Hopefully it will keep working correctly. Feel free to close this issue.

K4KDR, 2020-08-12:
Sincere thanks as always!

closed 2020-08-12 by K4KDR