daniestevez/gr-satellites #151: STRaND-1 Decoder

janvgils, 2020-08-16:
I wanted to decode STRaND-1 Telemetry and used GRIFEX.yml as an example, so created the following file.

STRAND-1.yml

name: STRAND-1
norad: 39090
data:
  &tlm Telemetry:
    telemetry: ax25
transmitters:
  9k6 FSK downlink:
    frequency: 437.568e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm

Telemetry example: https://network.satnogs.org/observations/2699592/

daniestevez, 2020-08-16:
Hi Jan, Thank you! Will you be submitting a pull request with this file or should I add it myself if that's easier for you?

janvgils, 2020-08-16:
Hi Dani,

I am not comfortable with all the git functions yet, so If you can, please do. 
Looking at other satellites with similar characteristics, so I really need to get familiar with git.

daniestevez, 2020-08-16:
No worries. I'll do this right away.

commit abf9bfa1 (2020-08-16): Add SatYAML file for STRAND-1

This closes #151
closed 2020-08-16 by daniestevez