daniestevez/gr-satellites #152: BY70-2 issue

ik8ozv, 2020-08-17:
Hello All,
Please anybody can you help me with the BY70-2 satellite?
I cannot understand why I receive data when the satellite is not transmitting
While when it transmits I do not get any data.
The command that I give in the terminals is:

gr_satellites BY02 --udp --samp_rate 48e3 --hexdump --kiss_out /home/enzo/Scrivania/BY70-2.kss

When I give the above command, I immediately get this answer:

Could not connect to telemetry proxy: Connection refused
Disabling telemetry submission...

Is it normal?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-17:
Duplicate of #147

daniestevez, 2020-08-17:
This is almost the same question as https://github.com/daniestevez/gr-satellites/issues/147#issuecomment-671035041, which is in a issue which is still open, so please re-use that issue for additional comments instead of opening another one.

I think that I gave you a complete answer on that issue, however. As said, unless the `--syncword_threshold` parameter is set lower, you will typically see false decodes. And with this I'm closing this issue.

ik8ozv, 2020-08-17:
Please Dani with is the correct command to give? Just  --syncword_threshold or i need to add something?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-17:
I think that either just trying it out or using `gr_satellites BY02 --help` will show you the correct syntax, so please put some effort on your side.

ik8ozv, 2020-08-17:
I have tried to add the command so i give the command  like this:

gr_satellites BY02 --udp --samp_rate 48e3 --syncword_threshold --hexdump --kiss_out /home/enzo/Scrivania/BY70-2.kss

and get this:

Could not connect to telemetry proxy: Connection refused
Disabling telemetry submission...
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff

Is it normal?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-08-17:
Well, no. That's not normal. I guess you have written the command incorrectly here, because I get the following:

```
$ gr_satellites BY02 --udp --samp_rate 48e3 --syncword_threshold --hexdump --kiss_out /home/enzo/Scrivania/BY70-2.kss
usage: gr_satellites satellite [-h] [--version]
                               (--wavfile WAVFILE | --rawfile RAWFILE | --rawint16 RAWINT16 | --udp | --kiss_in KISS_IN)
                               [--samp_rate SAMP_RATE] [--udp_ip UDP_IP]
                               [--udp_port UDP_PORT] [--iq]
                               [--input_gain INPUT_GAIN] [--kiss_out KISS_OUT]
                               [--kiss_append] [--hexdump]
                               [--dump_path DUMP_PATH]
                               [--telemetry_output TELEMETRY_OUTPUT]
                               [--codec2_ip CODEC2_IP]
                               [--codec2_port CODEC2_PORT]
                               [--f_offset F_OFFSET] [--rrc_alpha RRC_ALPHA]
                               [--disable_fll] [--fll_bw FLL_BW]
                               [--clk_bw CLK_BW] [--clk_limit CLK_LIMIT]
                               [--costas_bw COSTAS_BW]
                               [--manchester_history MANCHESTER_HISTORY]
                               [--syncword_threshold SYNCWORD_THRESHOLD]
gr_satellites satellite: error: argument --syncword_threshold: expected one argument
```

ik8ozv, 2020-08-17:
apologies why is incorrect?
maybe it should be written in block capitals?

daniestevez, 2020-08-17:
As you can see, when you run

```
gr_satellites BY02 --udp --samp_rate 48e3 --syncword_threshold --hexdump --kiss_out /home/enzo/Scrivania/BY70-2.kss
```

you don't get

```
Could not connect to telemetry proxy: Connection refused
Disabling telemetry submission...
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
```

Just try it for yourself.

PS: Since you're continuing the discussion here even though I marked this issue as duplicate and closed it immediately, I'll proceed to open this and close the other one.

ik8ozv, 2020-08-17:
Dani as in the help it says that syncword_threshold= 4, i have tried just with the command:

gr_satellites BY02 --udp --samp_rate 48e3 --syncword_threshold 4

but i get as before that you said it is not normal:

Could not connect to telemetry proxy: Connection refused
Disabling telemetry submission...
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 1acffc1d
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff

So which is the correct one then please?
Thanks

K4KDR, 2020-08-17:
>>>  Could not connect to telemetry proxy: Connection refused

This message appears if the Harbin "Mun Downlink Proxy" is NOT running and can be ignored.  Other functionality of the gr-satellites solution will function normally if a good signal is received via UDP audio.

daniestevez, 2020-08-17:
Ah, now you've pasted correctly here both the command you used and its output.

When I run this, I get

```
 $ gr_satellites BY02 --udp --samp_rate 48e3 --syncword_threshold 4
Could not connect to telemetry proxy: Connection refused
Disabling telemetry submission...
```

Your output and mine are both correct. And this may be a correct way to run gr_satellites, depending on what you want to achieve.

So what are those `gr::log` lines, you may ask. Those GNU Radio logging. Fully customizable to your liking. See [this](https://wiki.gnuradio.org/index.php/Logging).

ik8ozv, 2020-08-17:
Dani please attached is a short video that shows my problem.
How you can see while there is no signal from the bird, you can see incoming data in the terminal, while when i get the signal ( you can see it in the waterfall ) there is no incoming data .
Please can you tell me why this?
Thanks
Enzo
[BY70-2.zip](https://github.com/daniestevez/gr-satellites/files/5086818/BY70-2.zip)

K4KDR, 2020-08-17:
Demod BPSK with USB, not FM.

ik8ozv, 2020-08-17:
Apologise Scott what you mean?
Thanks
Enzo

K4KDR, 2020-08-17:
When you wish to decode a BPSK signal, your radio (or SDR software) should be set to USB.  When a packet is seen, adjust the tuning frequency to CENTER the packet and track for doppler:  

[image]

ik8ozv, 2020-08-18:
Scott,
you mean i have to set the modulation in GQRX to USB??
and what about the command to give in  terminal?
Always the same or it should be different?
Thanks a lot
73's de Enzo IK8OZV

K4KDR, 2020-08-18:
I can only imagine that random noise is producing some false decodes.  I get that sometimes, too.

And yes, set demod to USB as shown in my previous image.  That should change nothing in the decoder, but since I use the GUI in 'GNU Radio Companion', I am not as familiar with using the command line options.  So I cannot help there.

daniestevez, 2020-08-18:
The syncword threshold is changed using the `--syncword_threshold` parameter in `gr_satellites`. This does the same as the "Syncword threshold" field in the deframer blocks in GNU Radio Companion. The default value of 4 will give some false decodes with BY02, so you may use a lower value, such as `--syncword_threshold 0`, to reduce the number of false decodes.

ik8ozv, 2020-08-30:
Please anybody can tell me why i do not get any connected channel? 

[image]

Where am i missing?
Thanks

daniestevez, 2020-08-30:
That seems an issue with the telemetry proxy itself, which is part of [gr-lilacsat](https://github.com/bg2bhc/gr-lilacsat), not of gr-satellites.

K4KDR, 2020-08-30:
Be sure that the TCP ports are the same in GNU Radio & on the proxy.

[image]

[image]

ik8ozv, 2020-08-30:
Yes Scott it is
Thanks

daniestevez, 2020-09-28:
I'm closing this, since it's being open for a month without further discussion.

closed 2020-09-28 by daniestevez

ik8ozv, 2020-09-28:
I still have the problem.
Or better i can get the channel 5 connected but i do not see any incoming data in the Proxy.
Any advices please?
Thanks
73's de Enzo IK8OZV