daniestevez/gr-satellites #153: Chance for SoundCard Audio source input in Command Line ?

EA4SG, 2020-08-24:
Hi Daniel:

Is it possible to have input from default sound card input when ussing GR_Satellites from the Miniconda3 command line in Windows?

I see this is possible ussing your satellite decoder blocks in the GNU Radio blocks platform.

Thanks

EA4SG

daniestevez, 2020-08-24:
Hi David,

gr-satellites doesn't currently support reading audio from the sound card directly. However, it is possible to use the `audio_streamer` from [gr-frontends](https://github.com/daniestevez/gr-frontends) together with gr-satellites `--udp` input. The audio streamer is a simple GNU Radio flowgraph that reads from the sound card and streams samples using UDP, in the same format that gr-satellites uses. The audio streamer should work correctly in Windows, so please me let me know if you find any problems.

In any case, I will be leaving this issue open as a reminder to implement a soundcard input directly in a future version of gr-satellites.

EA4SG, 2020-08-25:
Thanks for your answer Daniel and thanks in advance for the future implementation.

When trying to open any gr-frontends grc files in gnuradio-companion, I get these errors:

Loading: "C:\Users\usuario\rtlsdr_fm.grc"
Error: mapping values are not allowed here
  in "C:\Users\usuario\rtlsdr_fm.grc", line 132, column 34
>>> Failure

 or 

Loading: "C:\Users\usuario\audio_streamer.grc"
Error: mapping values are not allowed here
  in "C:\Users\usuario\audio_streamer.grc", line 132, column 34
>>> Failure

Can you give me a clue where the fault may be?
Not very familiar with miniconda3 and gnuradio yet. Could be an issue due to a wrong file allocation (wrong directory) or something like that?
Or should I install gr-frontends from the Anaconda prompt?

In the maintime i am ussing a block called "Audio Source" located in the Core/Audio/ folder with certain success. I dont know if your suggestion (Audio streamer) will bring me better results or not

(miniconda3 + GNU Radio Companion 3.8.1.0 + Python 3.8.3 here)

73s de David

daniestevez, 2020-08-25:
Hi David,

I tried to replicate your problem by doing a fresh install of miniconda, gnuradio and gr-satellites in a Windows 10 machine. I can open the `audio_streamer.grc` in gnuradio-companion and it seems to run without any problems. In particular, I don't get the same error as you, which I find very strange.

By the way, I needed to enable the sound input device on Windows, in Settings > System > Sound. No idea why it was disabled on this machine.

daniestevez, 2020-08-29:
The `--audio` command line option to read samples from the audio soundcard has now been implemented in aebf8410dd3bfeeb764df1c0e3f10f164de2f2d1.

I will keep this issue open in case there is more information about the problems with gr-frontends in Windows.

daniestevez, 2020-09-13:
gr-satellites v3.4.0, which includes the `--audio` option to read directly from the soundcard is now avaliable in conda, so I'm closing this issue.

closed 2020-09-13 by daniestevez