daniestevez/gr-satellites #154: Installation through MacPorts succeeds, but crashes the Gnuradio when used

Sedenions, 2020-08-27:
Hi Daniel,

I can install the gr-satellites and gnuradio through fresh MacPort on the Catalina 10.15.6 macOS without any problems.

```
sudo port install gnuradio
sudo port install gr-satellites
```

But, when I start the gnu radio-companion, I got the following errors:

```
>>> Done
Warning: restarting the docstring loader (crashed while loading 'satellites_aausat4_deframer')
Warning: restarting the docstring loader (crashed while loading 'satellites_aausat4_check_fsm')
Warning: restarting the docstring loader (crashed while loading 'satellites_adsb_kml')
Warning: restarting the docstring loader (crashed while loading 'satellites_afsk_demodulator')
Warning: restarting the docstring loader (crashed while loading 'satellites_ao40_fec_deframer')
Warning: docstring loader crashed too often
```

By dragging any block from gr-satallites to graphical editor, the gnuradio crashes immediately.

**Temporary Solution**

Installation from source after removing gr-satellites from MacPort and keep all previously installed dependences on MacPort solves the issue completely.

```
sudo port uninstall gr-satellites

git clone https://github.com/daniestevez/gr-satellites
cd gr-satellites/
git checkout maint-3.8
mkdir build
cd build

cmake -DCMAKE_INSTALL_PREFIX=/opt/local ..
make
sudo make install

gnuradio-companion 
```

Now, the gnu radio starts without any errors, and blocks from gr-satellites can be dragged to editor without any problems.

```
(gnuradio-companion:66801): Gtk-WARNING **: 14:19:11.788: Locale not supported by C library.
	Using the fallback 'C' locale.
dbus[66801]: Dynamic session lookup supported but failed: launchd did not provide a socket path, verify that org.freedesktop.dbus-session.plist is loaded!
No such schema ?org.gnome.desktop.interface?
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/opt/local/share/gnuradio/grc/blocks

Loading: "/Users/Matjaz/Documents/gnuradio/FM.grc"
>>> Done
```

Thank you for Mac support. If you need anyone for testing the installation on Catalina macOS, I'm willing to do.

Best Regards, 
Matjaz - S59MZ

daniestevez, 2020-08-27:
Hi Matjaz,

Thanks for the report. I know almost nothing about MacPorts, so I don't understand well the difference between installing from MacPort and building from source manually. The error you get

> Warning: restarting the docstring loader (crashed while loading 'satellites_aausat4_deframer')

is also quite strange. It seems to speak about docstrings like [this one](https://github.com/daniestevez/gr-satellites/blob/b0d06d5751c7462887ae4f1ca4bcdd957740a853/grc/components/deframers/satellites_aausat4_deframer.block.yml#L28), but I don't know what it couldn't like about it. We would have to hunt in the gnuradio-companion code to see what it's doing when the error is produced.

Could you compare the `satellites_aausat4_deframer.block.yml` file (for instance) between the MacPort install that doesn't work and the manual install that does work to see if they are any different?

Sedenions, 2020-08-27:
I compared both files and they are the same. But, I've found something else. In both cases the Python files of mentioned blocks are located at Python 3.8 (!!) location: 

`./Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/satellites/components/deframers/aausat4_deframer.py
`

But, the MacPort installation log says about Python 3.7:

```
$ sudo port install gr-satellites
--->  Computing dependencies for gr-satellites
--->  Fetching archive for gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python37.darwin_19.x86_64.tbz2 from https://packages.macports.org/gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python37.darwin_19.x86_64.tbz2 from https://cph.dk.packages.macports.org/gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python37.darwin_19.x86_64.tbz2 from http://nue.de.packages.macports.org/gr-satellites
--->  Fetching distfiles for gr-satellites
--->  Verifying checksums for gr-satellites
--->  Extracting gr-satellites
--->  Configuring gr-satellites
--->  Building gr-satellites
--->  Staging gr-satellites into destroot
--->  Installing gr-satellites @3.3.0_0+docs+python37
--->  Activating gr-satellites @3.3.0_0+docs+python37
--->  Cleaning gr-satellites
--->  Scanning binaries for linking errors
--->  No broken files found.
--->  No broken ports found.
```

That means that I have only the Python version mismatch between gnuradio and gr-satellites. Both ports have also different variants available for installation.

I used the following commands to list all possible MacPorts variants:

```
port variants gnuradio
port variants gr-satellites
```

Those shows that the default [+] option for gnuradio is python38, which is wanted.
And default [+] option for gr-satellites is python37, which conflicts with installed gnuradio.

**Solution**

Install both MacPorts with python38 variant enabled:

```
sudo port install gnuradio +python38
sudo port install gr-satellites +python38
```

The installation log is now correct:

```
$ sudo port install gr-satellites +python38
--->  Computing dependencies for gr-satellites
--->  Fetching archive for gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python38.darwin_19.x86_64.tbz2 from https://packages.macports.org/gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python38.darwin_19.x86_64.tbz2 from https://cph.dk.packages.macports.org/gr-satellites
--->  Attempting to fetch gr-satellites-3.3.0_0+docs+python38.darwin_19.x86_64.tbz2 from http://nue.de.packages.macports.org/gr-satellites
--->  Fetching distfiles for gr-satellites
--->  Verifying checksums for gr-satellites
--->  Extracting gr-satellites
--->  Configuring gr-satellites
--->  Building gr-satellites
--->  Staging gr-satellites into destroot
--->  Installing gr-satellites @3.3.0_0+docs+python38
--->  Activating gr-satellites @3.3.0_0+docs+python38
--->  Cleaning gr-satellites
--->  Scanning binaries for linking errors
--->  No broken files found.
--->  No broken ports found.
```

The same variants should be also used for gr-osmosdr and gr-limesdr, if they're needed:

```
sudo port install gr-osmosdr +python38
sudo port install gr-limesdr +python38

```

That solves all issues about the _docstring_ loader crashes.

closed 2020-08-27 by Sedenions

daniestevez, 2020-08-28:
Great! Thanks for this research.

Sedenions, 2020-08-30:
No problem. I'm new on MacPorts too, so I learned something new :)