daniestevez/gr-satellites #156: Telemetry Parser options issue

Vinz87, 2020-08-29:
I'm trying to put together some components by myself to decode FACSAT-1 telemetry, following documentation and FACSAT-1 YAML file description, and I encounter the following error with the Telemetry Parser block:
```
Traceback (most recent call last):
  File "/Users/Vincenzo/Downloads/gqrx/a2.py", line 204, in <module>
    main()
  File "/Users/Vincenzo/Downloads/gqrx/a2.py", line 180, in main
    tb = top_block_cls()
  File "/Users/Vincenzo/Downloads/gqrx/a2.py", line 84, in __init__
    self.satellites_telemetry_parser_0 = satellites.components.datasinks.telemetry_parser('csp', options='')
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/satellites/components/datasinks/telemetry_parser.py", line 39, in __init__
    if options is not None and options.telemetry_output:
AttributeError: 'str' object has no attribute 'telemetry_output'
```

The problem seems to be that the Option parameter (which if I well understand should be the Command Line Options parameter in the GNURadio block) is "" by default, but this empty string is not treated as None in Python. So it looks for the "telemetry_output" attribute and it doesn't find it.
[untitled2.pdf](https://github.com/daniestevez/gr-satellites/files/5145345/untitled2.pdf)

daniestevez, 2020-08-29:
What version of gr-satellites are you using?

I think this problem was fixed in 897933745ceff6fb80835f6398aa26d737f30b0b, which was released is v3.3.0.

Vinz87, 2020-09-02:
It is indeed the same bug, and yes I was using v3.2.0, installed through MacPorts.
I upgrade to v3.3.0 and can confirm the bug is fixed, thank you.

closed 2020-09-02 by Vinz87