daniestevez/gr-satellites #157: Make test is failed

ugurcansoruc, 2020-08-29:
Hi Daniel, i get an error while testing the make test command, what could be the reason for this?

[image]

daniestevez, 2020-08-30:
Can you please run `ctest -V` to get a detailed output of the failures?

Also, please include what version of gr-satellites you're building.

ugurcansoruc, 2020-08-31:
I tried master and main-3.8, both got the same errors.
I am using version 3.8.2.0 (Python 3.6.9) of the gnu radio.
I ran the command Ctest -V, I added the errors I received in errorLog.txt file. 
[errorLog.txt](https://github.com/daniestevez/gr-satellites/files/5153122/errorLog.txt)

daniestevez, 2020-09-01:
This seems to indicate that the python module wasn't built properly.

To check if it is a problem of the test setup or it is a real problem, please install using `sudo make install` and `sudo ldconfig` and then run `import satellites` inside a `python3` interpreter.

ugurcansoruc, 2020-09-03:
I reinstalled using miniconda. Problem solved, thank you...

closed 2020-09-03 by daniestevez