daniestevez/gr-satellites #160: Problem while building in Ubuntu 20.04

Chayan2500, 2020-09-08:
[ 37%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/decode_rs_ccsds.c.o
[ 38%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/decode_rs_char.c.o
[ 40%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/encode_rs_8.c.o
[ 41%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/encode_rs_ccsds.c.o
[ 42%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/encode_rs_char.c.o
[ 44%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/taltab.c.o
[ 45%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/ccsds.c.o
[ 47%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/libfec/init_rs_char.c.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-satellites.so.v3.4-compat-xxx-xunknown'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:706: lib/CMakeFiles/gnuradio-satellites.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

daniestevez, 2020-09-08:
Please see about [liborc in the installation instructions](https://gr-satellites.readthedocs.io/en/latest/installation.html#dependencies)

daniestevez, 2020-11-09:
I'm closing this issue, since the problem seems to be that `liborc-0.4-dev` was missing, which is already documented in the instructions, and there have been no further comments in this issue. Feel free to re-open if the problem persists.

closed 2020-11-09 by daniestevez