daniestevez/gr-satellites #161: Can't install gr-satellites and gnuradio-dev

DL8LAQ, 2020-09-09:
Hi Daniel,

maybe you or another user can give me a hint to solve this problem. I spend hours searching the net for similar problems and a solution.

 73, Norbert

[Problems installing gr-satellites3.3.0.txt](https://github.com/daniestevez/gr-satellites/files/5196052/Problems.installing.gr-satellites3.3.0.txt)

daniestevez, 2020-09-09:
Dear Norbert,

Please include your report inside the message rather than as an attachment so that we can see it directly.

This seems a problem with your distribution rather than gr-satellites. Probably you can't build any other OOT modules from source. For some reason, `cmake` can't find GNU Radio when trying to build gr-satellites. The fact that it says `requested version 3.9` is strange. It should search version 3.8.

It seems you have a mixture of PPA packages and packages from focal, so maybe that's your problem.

daniestevez, 2020-09-09:
Yes. If you don't have anything particularly valuable in that install, a clean install of Ubuntu 20.04 would probably save headaches.

Choice of distribution depends a lot on taste and experience. I'm a happy user of Gentoo and Arch, and use Debian testing for the gr-satellites buildroot server. But I wouldn't necessarily recommend these to a beginner.

DL8LAQ, 2020-09-09:
The linux on my laptop is used only for gnuradio. I will do a clean install of 20.04 :-)

73, Norbert

DL8LAQ, 2020-09-10:
A clean install of Ubuntu 20.04 and installation of gnuradio-companion and gr_satellites using miniconda did it! 73, Norbert

closed 2020-09-10 by DL8LAQ