daniestevez/gr-satellites #165: File not found - compile_hierarchical.sh

XE3ADC, 2020-09-26:
Hello,
I performed the installation of gnuradio 3.8 and gr-satellites on a fresh ubuntu 18.04 LTS intallation (with MATE desktop) as per the following procedure:
https://github.com/daniestevez/gr-satellites/wiki/Installation-of-GNUradio-3.8-under-Ubuntu-18.04-LTS%2C-%28K%29Ubuntu-19.04%2C-Mint-19.2-Mate%2C-Mint-19.2-Cinnamon-or-Ubuntu-19.10

The procedure went well without problems with the exception of the following two Issues:
1-After the installation of gnuradio, when lauching gnuradio-companion to veryfy its version a popup window appeared displaying a "Module not found error".
2-After installation of gr-satellites the procedure asks to run the compile_hierarchical.sh script but when launched ubuntu responded that the file does not exist.

The first issue was solved by adding the following lines on ~/.profile file (As per https://stackoverflow.com/questions/60792169/modulenotfounderror-on-gnu-radio-right-after-installation):

export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3/dist-packages
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH

After applying this patch gnuradio-companion launches aparently well either in a terminal or from the graphical desktop icon.
As for the second issue, I could not locate the compile_hierarchical.sh script installed on the whole system, also could not find a similar report/solution on the web and ran out of ideas on how to proceed any further.
As mentioned before, gnuradio-companion seems to be executing fine, both the core and satellites modules appear on the right pane of the window, any ideas on what could be the cause of this issue?, please let me know how to proceed, any help will be greatly appreciated.
Thanks & Regards
Abraham - XE3ADC

daniestevez, 2020-09-26:
Hi,
The installation instructions you have been following are for an older version (v2) of gr-satellites. The installation instructions for the current version can be found [in the documentation](https://gr-satellites.readthedocs.io/en/latest/installation.html). As you can see, it is not needed to run `./compile_hierarchical.sh` any longer, so maybe your installation is working correctly already.

I have added a note in the Wiki pages that list instructions for older versions of gr-satellites. Maybe it would be better to delete them.

XE3ADC, 2020-09-26:
Many thanks Dani, I will continue with some testing and report back if required...
Best regards,
Abraham - XE3ADC

daniestevez, 2020-09-26:
Good. I will close this issue. Please open a new one if you find another problem.

closed 2020-09-26 by daniestevez