daniestevez/gr-satellites #167: 32-bit OS support

jgromes, 2020-09-30:
Hi,

I'm having some issues running gr-satellites on a 32-bit OS (Ubuntu 18.04). I downloaded all the dependencies, and it does compile  and install without issues. The problem is that when I try to run the command-line program, it crashes immediately with `Segmentation fault (core dumped)` - no help text, nothing. Opening in-terminal Python session and trying `import satellites` produces segfault too, so does running in a clean environment with only PYTHONPATH variable set.

The output from running `ctest -V` is the following (only first test shown, the others fail with the exact same error). I'm not sure if this is related to the segfault, or a different issue altogether.

```
UpdateCTestConfiguration  from :/home/gromes/repos/gr-satellites/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/gromes/repos/gr-satellites/build/DartConfiguration.tcl
Test project /home/gromes/repos/gr-satellites/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_hdlc

1: Test command: /bin/sh "/home/gromes/repos/gr-satellites/build/python/qa_hdlc_test.sh"
1: Test timeout computed to be: 10000000
1: Traceback (most recent call last):
1:   File "/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py", line 14, in swig_import_helper
1:     return importlib.import_module(mname)
1:   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
1:     return _bootstrap._gcd_import(name[level:], package, level)
1:   File "<frozen importlib._bootstrap>", line 994, in _gcd_import
1:   File "<frozen importlib._bootstrap>", line 971, in _find_and_load
1:   File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
1:   File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
1:   File "<frozen importlib._bootstrap>", line 571, in module_from_spec
1:   File "<frozen importlib._bootstrap_external>", line 922, in create_module
1:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
1: ImportError: /usr/local/lib/libgnuradio-satellites.so.3.4.0: undefined symbol: cpu_features
1: 
1: During handling of the above exception, another exception occurred:
1: 
1: Traceback (most recent call last):
1:   File "/usr/local/lib/python3/dist-packages/satellites/__init__.py", line 31, in <module>
1:     from .satellites_swig import *
1:   File "/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py", line 17, in <module>
1:     _satellites_swig = swig_import_helper()
1:   File "/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py", line 16, in swig_import_helper
1:     return importlib.import_module('_satellites_swig')
1:   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
1:     return _bootstrap._gcd_import(name[level:], package, level)
1: ImportError: /usr/local/lib/libgnuradio-satellites.so.3.4.0: undefined symbol: cpu_features
1: 
1: During handling of the above exception, another exception occurred:
1: 
1: Traceback (most recent call last):
1:   File "/home/gromes/repos/gr-satellites/build/swig/satellites_swig.py", line 14, in swig_import_helper
1:     return importlib.import_module(mname)
1:   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
1:     return _bootstrap._gcd_import(name[level:], package, level)
1:   File "<frozen importlib._bootstrap>", line 994, in _gcd_import
1:   File "<frozen importlib._bootstrap>", line 971, in _find_and_load
1:   File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
1:   File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
1:   File "<frozen importlib._bootstrap>", line 571, in module_from_spec
1:   File "<frozen importlib._bootstrap_external>", line 922, in create_module
1:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
1: ImportError: /usr/local/lib/libgnuradio-satellites.so.3.4.0: undefined symbol: cpu_features
1: 
1: During handling of the above exception, another exception occurred:
1: 
1: Traceback (most recent call last):
1:   File "/home/gromes/repos/gr-satellites/python/qa_hdlc.py", line 24, in <module>
1:     from satellites import hdlc_framer, hdlc_deframer
1:   File "/usr/local/lib/python3/dist-packages/satellites/__init__.py", line 33, in <module>
1:     from satellites_swig import *
1:   File "/home/gromes/repos/gr-satellites/build/swig/satellites_swig.py", line 17, in <module>
1:     _satellites_swig = swig_import_helper()
1:   File "/home/gromes/repos/gr-satellites/build/swig/satellites_swig.py", line 16, in swig_import_helper
1:     return importlib.import_module('_satellites_swig')
1:   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
1:     return _bootstrap._gcd_import(name[level:], package, level)
1: ImportError: /usr/local/lib/libgnuradio-satellites.so.3.4.0: undefined symbol: cpu_features
1/7 Test #1: qa_hdlc ..........................***Failed    0.65 sec
```

GNUradio does compile on this system and does work. I also have no issues running gr-satellites on a 64-bit Ubuntu 18.04. Is gr-satellites incompatible with a 32-bit system? Thanks, and also huge thanks for the amazing work you're doing here!

daniestevez, 2020-09-30:
Hi Jan, thanks for the report.

There is no reason why gr-satellites shouldn't work on any architecture, including i386, but I guess no one has tested if gr-satellites works on i386 since v3.3.0. I have realized that in that release I introduced a bug that doesn't show up in x86_64 but will cause your problem in i368. Continuous integration is done on an x86_64 box, so this didn't show up either.

I have just pushed a commit to master that should fix the problem. Could you please test with the current master?

jgromes, 2020-09-30:
Wow, that was quick, works perfectly now, both command line and from gnuradio - thanks!

BTW I'm working on decoders for a couple upcoming satellites, what's the preferred method of getting them into upstream - issue or PR? It's already implemented, working and tested, so I think it should be pretty straight-forward to just merge them from a PR.

closed 2020-09-30 by jgromes

daniestevez, 2020-09-30:
If you already have code, then a PR with the code would be appropriate.

Do these decoders fit into the demodulator-deframer-datasink framework in gr-satellites v3 and are usable from a SatYAML file? That would be the best way to ensure they are seamlessly integrated.

jgromes, 2020-09-30:
I based it on the 3.4.0 release, so yes - there's a custom deframer for one of the two, and telemetry decoders for both. The custom deframer introduced a couple new blocks for handling SX126x scrambler and CRC, but nothing major.

Will open a PR once it's all ready.

daniestevez, 2020-09-30:
That's good. More and more satellites are starting to fly the SX126x chips, so it's something I'll need to support one way or another. The last we've seen is NORBI, launched this Monday.

Regarding the scrambler: can't this be done with the GNU Radio Additive Scrambler block? That would be the preferred way to implement it.

jgromes, 2020-09-30:
> can't this be done with the GNU Radio Additive Scrambler block?

Probably, haven't checked. SX126x is using PN9 scrambling, but it's reversing bit order before the scrambler and after it, so I just reused the PN9 scrambler block and added new block for reversing bit order within one byte.

daniestevez, 2020-09-30:
I see, then maybe the Additive Scrambler alone can't be used if there is this bit reversal needed.

Note that there is already a `reflect_bytes` block in gr-satellites. I think this does the same as your bit order reversing block.

jgromes, 2020-09-30:
Missed that one - thanks!