daniestevez/gr-satellites #169: Use definitive NORAD IDs for satellites in 2020-068 Soyuz launch

daniestevez, 2020-09-30:
* [ ] SALSAT
* [ ] NETSAT 1-4
* [ ] MEZNSAT
* [ ] YARILO A, B (not yet in gr-satellite, unknown modulation)
* [ ] DEKART
* [ ] NORBI (not yet in gr-satellites; need more info about SX12xx settings)

N6RFM, 2020-09-30:
Great Progress here Dani!!  Thanks so much.  

For Norbi, are you referring to FSK only or do you also plan to try and add LoRa support as well?  My understanding is that gr-lora is likely inadequate and LoRa rx remains better served by direct use of sx126x/sx127x hardware.  I'm also not sure if there is a good way to interface such hardware to gr-satellites or gr-frontends (which would be wonderful if possible).  :-)

Cheers,
Bob

daniestevez, 2020-09-30:
I only plan to support FSK.

The last time I used gr-lora surprisingly it worked just fine. This was for the GRCon20 CTF, and I think the recording was a real off-the-air signal. Seeing the latest commits, it seems gr-lora has seen some improvement this year (I didn't study the specific details), so maybe it works much better than before and we should give it another opportunity.

So in case gr-lora works well for NORBI and the other LoRa satellites that are going to be launched, we can just use that, and if not the SX12xx chips.

The SX12xx chips are a complete hardware transceiver on their own, so it's not really possible to interface them to an SDR running on your computer. They expect to receive an RF LoRa signal. What is possible is to make an IQ recording with an SDR receiver and then replay that recording through RF (coax and attenuator) into the SX12xx. This can be used for several things: doing a wideband recording if you're not sure where the LoRa transmissions are going to appear, using a recording that you did without using the SX12xx because you didn't expect to find a LoRa signal, correcting for Doppler with the computer if the LoRa receiver can't handle all the Doppler, trying different SX12xx parameters, etc. In fact, most if not all the LoRa decodes that have been done with FossaSat-1 used this method of record and replay.

N6RFM, 2020-09-30:
Thanks Dani. QSL.  Will also try gr-lora again once we have an active LoRa signal to look at.  Indeed, I certainly recall the Fossasat-1 team using the record/replay approach.  Best,  Bob

daniestevez, 2020-10-02:
After some of PA0DLO's tweets, I'm assigning:
* NORBI 46494
* MEZNSAT 46489
* SALSAT 46495
These are most likely correct but will need to be checked in a few days.

daniestevez, 2020-10-08:
Now I'm setting
* DEKART 46493

daniestevez, 2020-10-17:
NETSAT 1-4 are still not identified, and the YARILOs don't have support in gr-satellites yet, so I'm pushing this to v3.6.0.

daniestevez, 2020-11-05:
The NETSATs have been identified in b272725f4e30f8d57925dc47aaf109355034e77c

I'm closing this, since YARILO A and B are not currently supported in gr-satellites, since I haven't found a good recording or any documentation about their transmissions.

closed 2020-11-05 by daniestevez