daniestevez/gr-satellites #170: Direct OGG file support?

N6RFM, 2020-09-30:
Hi Dani,

Seems SatNogs network station data stored as ogg format.  Easy to manually convert these to wav, but wondering is ogg format could be supported directly in future?  -- oggfile?

Thanks all efforts!!
Bob

daniestevez, 2020-09-30:
Hi Bob,

That's a very good question and something I also wonder myself. It would be quite convenient and save as all from having to convert this SatNOGS OGGs with `oggdec` or `sox`. On the other hand, I feel that maybe this should be done in GNU Radio rather than gr-satellites, since it's something general and other people could benefit from it.

In fact, there is some movement now to change the WAV file reading code in GNU Radio (which is what gr-satellites uses to read WAV files), because it doesn't even support all types of WAV files. Whether GNU Radio will end up supporting other formats such as OGG as part of this movement is perhaps still too soon to know.

But yes, this is something that is foreseen to be in gr-satellites in the future, one way or another.

daniestevez, 2020-09-30:
Ah, in fact now I managed to find the information I was looking for (formerly I had just heard this from other people):

https://github.com/gnuradio/gnuradio/pull/3600

So this will be in GNU Radio 3.9 and it supports FLAC and OGG.

Therefore, the reasonable option seems to be to wait for GNU Radio 3.9 and then just use that feature.

N6RFM, 2020-09-30:
Thanks Dani.  Sounds like a plan!  QRX for 3.9.  Best,  Bob

daniestevez, 2021-03-06:
Starting with GNU Radio 3.9, the WAV File Source block that is used by the `gr_satellites` `--wavfile` option uses libsndfile and supports OGG, FLAC and other formats. I have just pushed a commit that mentions this in the `--help` and in the docs. I have decided to maintain the name of the `--wavfile` option (rather than renaming it as `--sndfile` or `--audiofile`) because the GNU Radio block is still called WAV File Source and because of backwards compatibility.

I have tested that decoding OGGs from SatNOGS works. For instance, we can

```
wget https://network-satnogs.freetls.fastly.net/media/data_obs/3731739/satnogs_3731739_2021-03-06T10-49-34.ogg \
&& gr_satellites GRIFEX --wavfile satnogs_3731739_2021-03-06T10-49-34.ogg
```

I'm closing this issue.

closed 2021-03-06 by daniestevez