daniestevez/gr-satellites #171: --start_time  argument

K4KDR, 2020-09-30:
Using GRC on Ubuntu 18.04, 64-bit:

```
gr_satellites v3.5.0-git

heads/master-0-g2f09d5c
```

-----------------------------------------------------------------------------

Hoping that I understood the proper use of the recently added "--start_time" argument, I attempted to utilize it for replay of recorded data into the "Telemetry Submit" block as follows:

[image]

... unfortunately, the following is returned:

```
Executing: /usr/bin/python3.6 -u /home/k4kdr/grc/modified/kiss_submitter.py

usage: telemetry_submit [-h]
telemetry_submit: error: unrecognized arguments: --start_time 2020-09-30 17:58:25

>>> Done (return code 2)
```

... can this argument be used in the "Telemetry Submit" block?

Thank you!

daniestevez, 2020-09-30:
Hi Scott,

The problem is the space between the date and the time. This splits the date+time argument in two and the time is understood as a different argument. You need to use quotes:
``"--start_time '2020-09-30 17:58:25'"``
Also note that you need to play back the recording at 1x, so if you're using your own flowgraph you need to include a Throttle block (if using `gr_satellites` you can use the `--throttle` option).

PS: I tested this with `gr_satellites`, not directly with the Telemetry Submit block. I think it should work, but if it doesn't, please let me know.

K4KDR, 2020-09-30:
Unfortunately, the same error message is returned using that syntax.

daniestevez, 2020-10-01:
Hi Scott,
You're right. In fact this is a bug, but I need to think of a good way to fix it, as there isn't a straightforward solution. Will probably do so in the next few days.

K4KDR, 2020-10-01:
Thank you!

Yes, decodes from a recording replay are extremely common (at least for me) and I would like to contribute to the SatNogs DB as accurately as possible.

daniestevez, 2020-10-02:
I have pushed a new commit to `master` that I think will fix this problem (it wasn't as difficult as I thought). Please test it and feel free to close the issue if it works.

kerel-fs, 2020-10-02:
> The problem is the space between the date and the time. This splits the date+time argument in two and the time is understood as a different argument. You need to use quotes:
> `"--start_time '2020-09-30 17:58:25'"`

ISO-8601-compatible timestamps using a `T` instead of a white-space character (i.e. `2020-09-30T17:58:25`) would solve this issue. I hope the feature is young enough to allow for the syntax to change.

daniestevez, 2020-10-02:
Yes. The only reason why a space instead of a `T` was used is because that's the way that was implemented in gr-satellites v1, and I didn't touch the timestamp parsing code. To me it also makes more sense to use ISO-8601, so I'll change this right away.

K4KDR, 2020-10-02:
Thank you for updates!

The "--help" now indicates

```
--start_time START_TIME 
              Recording start timestamp
```

... so do I understand from the last few posts here that the syntax is now

`--start_time YYYY-MM-DDTHH:MM:SS`

??

daniestevez, 2020-10-02:
Actually both `YYYY-MM-DDTHH:MM:SS` and `'YYYY-MM-DD HH:MM:SS'` should work now, but if you use the space as separator you must include the enclosing quotes.

daniestevez, 2020-10-05:
Hi all, I'm coming back to this to see if this bug can be closed.

Is it working correctly for you now @K4KDR?

K4KDR, 2020-10-05:
Using MASTER > 1ffca19

Attempting to run using either syntax  (YYYY-MM-DDTHH:MM:SS or 'YYYY-MM-DD HH:MM:SS') in the "TELEMETRY SUBMIT" block  produces:

```
Traceback (most recent call last):
  File "/home/k4kdr/grc/modified/kiss_submitter.py", line 168, in <module>
    main()
  File "/home/k4kdr/grc/modified/kiss_submitter.py", line 144, in main
    tb = top_block_cls(host=options.host, port=options.port)
  File "/home/k4kdr/grc/modified/kiss_submitter.py", line 82, in __init__
    self.satellites_telemetry_submit_0 = satellites.components.datasinks.telemetry_submit("SatNOGS", norad=46489, port='0', config=satellites.utils.config.open_config(), options="--start_time 2020-10-01T23:25:00")
  File "/usr/local/lib/python3/dist-packages/satellites/components/datasinks/telemetry_submit.py", line 44, in __init__
    initial_timestamp)
  File "/usr/local/lib/python3/dist-packages/satellites/submit.py", line 35, in __init__
    if initialTimestamp != '' else None
AttributeError: type object 'datetime.datetime' has no attribute 'fromisoformat'

>>> Done (return code 1)
```

daniestevez, 2020-10-05:
The problem is that `fromisoformat()` was introduced in Python 3.7. I guess you're using an older version of Python.

I have pushed a new commit that does date parsing in another way if `fromisoformat()` is not available, so that this can also work with older versions of Python 3. However, in this case it is mandatory to use `T` as separator between the date and time.

K4KDR, 2020-10-05:
Thank you!  The flowgraph now runs without error using a "--start-time" argument.  I don't have data to test with at the moment, but will do so first chance!

daniestevez, 2020-10-17:
I've done some tests and this seems to be working. If you run into problems again, please feel free to re-open.

closed 2020-10-17 by daniestevez