daniestevez/gr-satellites #173: Decoding KAIDUN-1

janvgils, 2020-10-03:
Good afternoon all,

Just wanted to share my experiments to decode KAIDUN-1 that uses 1k2 BPSK G3RUH.

I am not completely sure if the output is correct, please share your thoughts.

Here the KAIDUN-1.yml file:

```
name: KAIDUN-1
norad: 41915
data:
  &tlm Telemetry:
    telemetry: ax25
transmitters:
  1k2 BPSK downlink:
    frequency: 437.600e+6
    modulation: BPSK
    baudrate: 1200
    framing: AX.25 G3RUH
    data:
    - *tlm
```
The command line that I used
```
gr_satellites 41915 --wavfile KAIDUN-1_satnogs_2879128_2020-09-22T15-03-35.wav --samp_rate 48e3 --f_offset 12.5e3

```
And the output it generates:
```
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'GSNWPU' (total 6)
                ssid = Container: 
                    ch = True
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'AXNWPU' (total 6)
                ssid = Container: 
                    ch = False
                    ssid = 4
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'Aoxiang star comes' (total 18)
```

daniestevez, 2020-10-05:
Hi Jan,

Thanks for sharing. That seems a good decode!

I'll add the KAIDUN-1.yml file to gr-satellites.

daniestevez, 2020-10-05:
Added in 1ffca193efe371c872a37450643b4c658b5b9ab4

closed 2020-10-05 by daniestevez