daniestevez/gr-satellites #174: Warnings when building the gr-satellites 3.5.0-git release on Debian Testing.

janvgils, 2020-10-08:
Some warning during the build process:

Are these indicators that should be fixed for coming releases, maybe some new dependencies ?

```
[ 50%] Generating documentation with doxygen
warning: Tag 'TCL_SUBST' at line 207 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'PERL_PATH' at line 1686 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1707 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[ 50%] Built target doxygen_target
Scanning dependencies of target _satellites_swig_doc_tag
[ 52%] Building CXX object swig/CMakeFiles/_satellites_swig_doc_tag.dir/_satellites_swig_doc_tag.cpp.o
[ 53%] Linking CXX executable _satellites_swig_doc_tag
[ 53%] Built target _satellites_swig_doc_tag
Scanning dependencies of target satellites_swig_swig_doc
[ 54%] Generating doxygen xml for satellites_swig_doc docs
warning: Tag 'TCL_SUBST' at line 207 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'PERL_PATH' at line 1654 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1675 of file '/home/janvg/source/git/daniestevez/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: argument 'a4wide' for option PAPER_TYPE is not a valid enum value
Using the default: a4!
[ 56%] Generating python docstrings for satellites_swig_doc
[ 56%] Built target satellites_swig_swig_doc
Scanning dependencies of target satellites_swig_swig_compilation
[ 57%] Swig source satellites_swig.i
[ 57%] Built target satellites_swig_swig_compilation
Scanning dependencies of target satellites_swig
[ 58%] Building CXX object swig/CMakeFiles/satellites_swig.dir/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx.o
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function ‘PyTypeObject* SwigPyObject_TypeOnce()’:
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1770:5: warning: missing initializer for member ‘_typeobject::tp_vectorcall’ [-Wmissing-field-initializers]
 1770 |     };
      |     ^
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1770:5: warning: missing initializer for member ‘_typeobject::tp_print’ [-Wmissing-field-initializers]
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function ‘PyTypeObject* SwigPyPacked_TypeOnce()’:
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1931:5: warning: missing initializer for member ‘_typeobject::tp_vectorcall’ [-Wmissing-field-initializers]
 1931 |     };
      |     ^
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1931:5: warning: missing initializer for member ‘_typeobject::tp_print’ [-Wmissing-field-initializers]
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function ‘PyTypeObject* swig_varlink_type()’:
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:65013:7: warning: missing initializer for member ‘_typeobject::tp_vectorcall’ [-Wmissing-field-initializers]
65013 |       };
      |       ^
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:65013:7: warning: missing initializer for member ‘_typeobject::tp_print’ [-Wmissing-field-initializers]
```

daniestevez, 2020-10-08:
Hi Jan,

Thanks for the report. I also get the warnings when the Doxygen documentation is generated. They are not very important, since Sphinx is what is used to generate the used documentation. The Doxygen documentation exists as automatic documentation for any GNU Radio OOT module, and I don't pay much attention to it for gr-satellites (though it would be something to improve, to document the C++ code better).

I don't get the warnings regarding SWIG when building on my system, and I don't know what they are about. I will ignore them. Hopefully they won't turn into something more serious.

So answering your question, really nothing that should fix on your side.

daniestevez, 2020-10-16:
I'm closing this, since the warnings are nothing to worry about.

closed 2020-10-16 by daniestevez