daniestevez/gr-satellites #176: Runtime error with latest commit 0a5a7ee

jharwinbarrozo, 2020-10-10:
$ gr_satellites FUNcube-1 --wavfile ./satellite-recordings/ao73.wav --samp_rate 48e3 
Could not find port: out in:
system

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 214, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 200, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 107, in __init__
    self.flowgraph = satellites.core.gr_satellites_flowgraph(samp_rate = options.samp_rate, iq = self.options.iq,
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 164, in __init__
    deframer = self.get_deframer(transmitter['framing'])(options = options, **deframer_additional_options)
  File "/usr/local/lib/python3/dist-packages/satellites/components/deframers/ao40_fec_deframer.py", line 53, in __init__
    self.msg_connect((self.deframer, 'out'), (self.deinterleaver, 'in'))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 60, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 132, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/runtime_swig.py", line 3006, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

Git checking out to 3.5.0 release https://github.com/daniestevez/gr-satellites/commit/b8842b3ce77a93363180336ed93f7151a373de9c solves the issue.

5ch4um1, 2020-10-10:
Which version of gnuradio are you using, and how did you install it?
I just followed the install instructions and could not reproduce your error:

$ gr_satellites FUNcube-1 --wavfile ./satellite-recordings/ao73.wav --samp_rate 48e3
Frame type WO10
----------------------------------------
Realtime telemetry:
----------------------------------------
Container: 
    eps = Container: 
        photovoltage = ListContainer: 
            0
            0
            0
        photocurrent = 0
        batteryvoltage = 8140
        systemcurrent = 206
        rebootcount = 721
        softwareerrors = 0
        boostconvertertemp = ListContainer: 

and so on...

$ gr_satellites --version
gr_satellites v3.6.0-git

daniestevez, 2020-10-10:
I can't reproduce the error either, and I can't think of any reason for the changes between https://github.com/daniestevez/gr-satellites/commit/b8842b3ce77a93363180336ed93f7151a373de9c and https://github.com/daniestevez/gr-satellites/commit/0a5a7ee33082ceb471c44a4343c8c5e40a2be161 to cause this.

jharwinbarrozo, 2020-10-11:
I installed my gnuradio via ubuntu ppa installation, i believe it is version 3.8.2
`sudo add-apt-repository ppa:gnuradio/gnuradio-releases`
`sudo apt-get update`
`sudo apt install gnuradio`

daniestevez, 2020-10-11:
What does `ctest -V` inside the `build` directory report with the commit that doesn't work for you?

jharwinbarrozo, 2020-10-11:
> What does `ctest -V` inside the `build` directory report with the commit that doesn't work for you?

Hi Daniel,

Here's the full log from cmake --> make ---> make test > ctest -V

jharwin@jharwin-ubuntu://home/jharwin/test/gr-satellites/$ git log -1
commit 0a5a7ee33082ceb471c44a4343c8c5e40a2be161 (HEAD -> master, origin/master, origin/HEAD)
Author: Daniel Estévez <daniel@destevez.net>
Date:   Thu Oct 8 19:54:07 2020 +0200

    Update NORAD ID for DEKART
    
    See #169
jharwin@jharwin-ubuntu:/home/jharwin/test/gr-satellites$ mkdir build
jharwin@jharwin-ubuntu:/home/jharwin/test/gr-satellites$ cd build
jharwin@jharwin-ubuntu/home/jharwin/test/gr-satellites/build$ cmake ..
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.5") 
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Using install prefix: /usr/local
-- Building for version: v3-alpha4-294-g0a5a7ee3 / 3.6.0git
-- No C++ unit tests... skipping
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found SWIG: /usr/bin/swig4.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jharwin/test/gr-satellites/build
jharwin@jharwin-ubuntu:/home/jharwin/test/gr-satellites/build$ make
Scanning dependencies of target gnuradio-satellites
[  1%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_impl.cc.o
[  2%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/encode_rs_impl.cc.o
[  4%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ao40_rs_decoder_impl.cc.o
[  5%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ax100_decode_impl.cc.o
[  6%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/u482c_decode_impl.cc.o
[  8%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/randomizer.c.o
[  9%] Building C object lib/CMakeFiles/gnuradio-satellites.dir/golay24.c.o
[ 10%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/lilacsat1_demux_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/varlen_packet_tagger_impl.cc.o
[... 225 lines cut ...]
    Start 6: qa_satyaml

6: Test command: /usr/bin/sh "/home/jharwin/test/gr-satellites/build/python/satyaml/qa_satyaml_test.sh"
6: Test timeout computed to be: 10000000
6: .
6: ----------------------------------------------------------------------
6: Ran 1 test in 0.192s
6: 
6: OK
6/7 Test #6: qa_satyaml .......................   Passed    0.26 sec
test 7
    Start 7: qa_by02

7: Test command: /usr/bin/sh "/home/jharwin/test/gr-satellites/build/python/telemetry/qa_by02_test.sh"
7: Test timeout computed to be: 10000000
7: .
7: ----------------------------------------------------------------------
7: Ran 1 test in 0.000s
7: 
7: OK
7/7 Test #7: qa_by02 ..........................   Passed    0.23 sec

100% tests passed, 0 tests failed out of 7

Total Test time (real) =   1.98 sec

daniestevez, 2020-10-11:
So all your tests seem to pass, including `qa_ao40_fec_deframer_test`, which is a test of the AO-40 FEC deframer block. Interestingly, the failure you got when running `gr_satellites FUNcube-1 --wavfile ./satellite-recordings/ao73.wav --samp_rate 48e3` was in the constructor of `ao40_fec_deframer`. But the test is running properly. I wonder what is happening.

Maybe you have a mixture of installations on your system, if you've been installing different versions of gr-satellites in your system. Perhaps it would be good to remove all trace of gr-satellites on your system and install again. The test runs directly from the `build` directory, so it wouldn't be affected by this.

jharwinbarrozo, 2020-10-12:
When I installed the https://github.com/daniestevez/gr-satellites/commit/0a5a7ee33082ceb471c44a4343c8c5e40a2be161 I had it on a fresh ubuntu 20.04 clean slate, so I am pretty sure I don't have any gr-satellites installed anywhere. 

The directory where I ran the ctest -v was just a new git clone from the latest commit, but I didn't install it(sudo make install).

I am going to close this as I don't have any idea too what causes it, as the test were all okay.

closed 2020-10-12 by jharwinbarrozo