daniestevez/gr-satellites #180: Make Errors - commit/960a9edf124a97e43a214bf42327c0ed220cc46c

N6RFM, 2020-10-14:
Hi Dani,

Just did a fresh install of the latest commit. 960a9ed 
Seeing several make test related errors.  Results OK for earlier commit - see below.
Any suggestions? Thanks as always!
Bob
N6RFM

Running tests...
Test project /home/bob/gr-satellites/build
    Start 1: qa_hdlc
1/8 Test #1: qa_hdlc ..........................   Passed    0.77 sec
    Start 2: qa_kiss
2/8 Test #2: qa_kiss ..........................   Passed    0.74 sec
    Start 3: qa_nrzi
3/8 Test #3: qa_nrzi ..........................   Passed    0.74 sec
    Start 4: qa_pdu_head_tail
4/8 Test #4: qa_pdu_head_tail .................***Failed    0.78 sec
    Start 5: qa_ao40_fec_deframer
5/8 Test #5: qa_ao40_fec_deframer .............   Passed    1.21 sec
    Start 6: qa_telemetry_parser
6/8 Test #6: qa_telemetry_parser ..............***Failed    0.76 sec
    Start 7: qa_satyaml
7/8 Test #7: qa_satyaml .......................   Passed    0.33 sec
    Start 8: qa_by02
8/8 Test #8: qa_by02 ..........................***Failed    0.08 sec

63% tests passed, 3 tests failed out of 8

Total Test time (real) =   5.43 sec

The following tests FAILED:
	  4 - qa_pdu_head_tail (Failed)
	  6 - qa_telemetry_parser (Failed)
	  8 - qa_by02 (Failed)
Errors while running CTest
Makefile:85: recipe for target 'test' failed
make: *** [test] Error 8
bob@bob:~/gr-satellites/build$ ^C

Previous commit tested OK

bob@bob:~/gr-satellites-479dcd4e62d36adf106fb5fd1418a2848ee72b9b/build$ make test
Running tests...
Test project /home/bob/gr-satellites-479dcd4e62d36adf106fb5fd1418a2848ee72b9b/build
    Start 1: qa_hdlc
1/7 Test #1: qa_hdlc ..........................   Passed    0.72 sec
    Start 2: qa_kiss
2/7 Test #2: qa_kiss ..........................   Passed    0.68 sec
    Start 3: qa_nrzi
3/7 Test #3: qa_nrzi ..........................   Passed    0.69 sec
    Start 4: qa_ao40_fec_deframer
4/7 Test #4: qa_ao40_fec_deframer .............   Passed    1.13 sec
    Start 5: qa_telemetry_parser
5/7 Test #5: qa_telemetry_parser ..............   Passed    0.70 sec
    Start 6: qa_satyaml
6/7 Test #6: qa_satyaml .......................   Passed    0.33 sec
    Start 7: qa_by02
7/7 Test #7: qa_by02 ..........................   Passed    0.36 sec

100% tests passed, 0 tests failed out of 7

Total Test time (real) =   4.63 sec
bob@bob:~/gr-satellites-479dcd4e62d36adf106fb5fd1418a2848ee72b9b/build$

daniestevez, 2020-10-14:
Thanks for the report!

I just checked that the tests run OK for me (and the continuous integration also does this for me automatically; this is what puts a green tick next to the commits), so I wonder why they are failing for you.

Can you please run `ctest -V` inside the `build` folder? In fact, just to be sure that nothing left over in the `build` directory is disturbing, you can run from the `gr-satellites` directory
```
rm -rf build && mkdir build && cd build && cmake .. && make -j4 && ctest -V
```

N6RFM, 2020-10-14:
Thanks Dani.  Still a puzzle.

Here is output from clean git clone, then rm -rf build && mkdir build && cd build && cmake .. && make -j4 && ctest -V

[100%] Built target pygen_swig_ba5ab
UpdateCTestConfiguration  from :/home/bob/gr-satellites/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/bob/gr-satellites/build/DartConfiguration.tcl
Test project /home/bob/gr-satellites/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_hdlc

1: Test command: /bin/sh "/home/bob/gr-satellites/build/python/qa_hdlc_test.sh"
1: Test timeout computed to be: 9.99988e+06
1: .
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.014s
1: 
1: OK
1/8 Test #1: qa_hdlc ..........................   Passed    0.82 sec
test 2
    Start 2: qa_kiss

2: Test command: /bin/sh "/home/bob/gr-satellites/build/python/qa_kiss_test.sh"
2: Test timeout computed to be: 9.99988e+06
2: .
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.005s
2: 
2: OK
2/8 Test #2: qa_kiss ..........................   Passed    0.69 sec
test 3
    Start 3: qa_nrzi

3: Test command: /bin/sh "/home/bob/gr-satellites/build/python/qa_nrzi_test.sh"
3: Test timeout computed to be: 9.99988e+06
3: ...
3: ----------------------------------------------------------------------
3: Ran 3 tests in 0.003s
3: 
3: OK
3/8 Test #3: qa_nrzi ..........................   Passed    0.69 sec
test 4
    Start 4: qa_pdu_head_tail

4: Test command: /bin/sh "/home/bob/gr-satellites/build/python/qa_pdu_head_tail_test.sh"
4: Test timeout computed to be: 9.99988e+06
4: Traceback (most recent call last):
4:   File "/home/bob/gr-satellites/python/qa_pdu_head_tail.py", line 24, in <module>
4:     from satellites import pdu_head_tail
4: ImportError: cannot import name 'pdu_head_tail'
4/8 Test #4: qa_pdu_head_tail .................***Failed    0.75 sec
test 5
    Start 5: qa_ao40_fec_deframer

5: Test command: /bin/sh "/home/bob/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
5: Test timeout computed to be: 9.99988e+06
5: .
5: ----------------------------------------------------------------------
5: Ran 1 test in 0.019s
5: 
5: OK
5/8 Test #5: qa_ao40_fec_deframer .............   Passed    1.14 sec
test 6
    Start 6: qa_telemetry_parser

6: Test command: /bin/sh "/home/bob/gr-satellites/build/python/components/datasinks/qa_telemetry_parser_test.sh"
6: Test timeout computed to be: 9.99988e+06
6: E
6: ======================================================================
6: ERROR: test_instance (__main__.qa_telemetry_parser)
6: Tries to create a telemetry parser instance using several combinations of parameters
6: ----------------------------------------------------------------------
6: Traceback (most recent call last):
6:   File "/home/bob/gr-satellites/python/components/datasinks/qa_telemetry_parser.py", line 37, in test_instance
6:     telemetry_parser(definition, file = sys.stderr)
6: NameError: name 'sys' is not defined
6: 
6: ----------------------------------------------------------------------
6: Ran 1 test in 0.001s
6: 
6: FAILED (errors=1)
6/8 Test #6: qa_telemetry_parser ..............***Failed    0.70 sec
test 7
    Start 7: qa_satyaml

7: Test command: /bin/sh "/home/bob/gr-satellites/build/python/satyaml/qa_satyaml_test.sh"
7: Test timeout computed to be: 9.99988e+06
7: .
7: ----------------------------------------------------------------------
7: Ran 1 test in 0.251s
7: 
7: OK
7/8 Test #7: qa_satyaml .......................   Passed    0.31 sec
test 8
    Start 8: qa_by02

8: Test command: /bin/sh "/home/bob/gr-satellites/build/python/telemetry/qa_by02_test.sh"
8: Test timeout computed to be: 9.99988e+06
8: Segmentation fault (core dumped)
8/8 Test #8: qa_by02 ..........................***Failed    0.08 sec

63% tests passed, 3 tests failed out of 8

Total Test time (real) =   5.21 sec

The following tests FAILED:
	  4 - qa_pdu_head_tail (Failed)
	  6 - qa_telemetry_parser (Failed)
	  8 - qa_by02 (Failed)
Errors while running CTest
bob@bob:~/gr-satellites/build$

daniestevez, 2020-10-14:
I think I might have an idea of what is happening.

The tests are picking the version of gr-satellites that you have installed in the system (which might be a different one) rather than the one you've just built in the build folder. This would give all sorts of problems. I'm not sure how this could happen, though.

First, please try to install the new version of gr-satellites (just run `sudo make install` and `sudo ldconfig`) and then try running `ctest -V` again. If this works, I'm still quite curious about why the tests are picking gr-satellites from the wrong place, so it might be good to remove gr-satellites from the system (`sudo make uninstall`) and then running `ctest -V`. Probably they'll give a meaningful error message.

> 6: telemetry_parser(definition, file = sys.stderr)
6: NameError: name 'sys' is not defined

This is a potential problem that I've just fixed in be3cd8268d13371b167e3b40b0554db196da0bce.

N6RFM, 2020-10-14:
All fixed!  Thank you! sudo make install then sudo ldconfig.  Then make test or cmake -V give passing results. 
Yes, I was hopscotching versions to troubleshoot  What surprised me is that I never ran sudo make install.  Just make test and saw the error.   
Test project /home/bob/gr-satellites/build
    Start 1: qa_hdlc
1/8 Test #1: qa_hdlc ..........................   Passed    0.74 sec
    Start 2: qa_kiss
2/8 Test #2: qa_kiss ..........................   Passed    0.75 sec
    Start 3: qa_nrzi
3/8 Test #3: qa_nrzi ..........................   Passed    0.72 sec
    Start 4: qa_pdu_head_tail
4/8 Test #4: qa_pdu_head_tail .................   Passed    0.73 sec
    Start 5: qa_ao40_fec_deframer
5/8 Test #5: qa_ao40_fec_deframer .............   Passed    1.18 sec
    Start 6: qa_telemetry_parser
6/8 Test #6: qa_telemetry_parser ..............   Passed    0.74 sec
    Start 7: qa_satyaml
7/8 Test #7: qa_satyaml .......................   Passed    0.35 sec
    Start 8: qa_by02
8/8 Test #8: qa_by02 ..........................   Passed    0.38 sec

100% tests passed, 0 tests failed out of 8

Total Test time (real) =   5.59 sec

daniestevez, 2020-10-15:
Thanks! Good that it's working.

closed 2020-10-15 by daniestevez