daniestevez/gr-satellites #181: Build error with boost >= 1.73: `_1` placeholders not globally defined

ryanvolz, 2020-10-16:
As with https://github.com/gnuradio/gnuradio/issues/3720 in gnuradio, changes are needed to make gr-satellites compatible with boost >= 1.73. For example:
```
lib/decode_rs_impl.cc:45:84: error: '_1' was not declared in this scope
     set_msg_handler(pmt::mp("in"), boost::bind(&decode_rs_impl::msg_handler, this, _1));
                                                                                    ^~
```

See also [here](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=226169&view=logs&jobId=bb1c2637-64c6-57bd-9ea6-93823b2df951&j=bb1c2637-64c6-57bd-9ea6-93823b2df951&t=350df31b-3291-5209-0bb7-031395f0baa1) for failed build output.

GNU Radio looks to be moving to lambda functions over boost::bind and the placeholders in general, so that is probably the path to take here.

daniestevez, 2020-10-16:
I have just pushed a commit replacing `boost::bind()` by lambdas as GNU Radio is doing in the master branch (interestingly in the maint-3.8 branch they still are using `boost::bind()`).

I'm using boost 1.74.0 in my Gentoo system, but I had no problems building gr-satellites because the way that I solve the placeholders problems when building GNU Radio is to add
```
#include <boost/bind/placeholders.hpp>
using namespace boost::placeholders;
```
in some of the GNU Radio `.h` files, which basically gets included everywhere, including OOT modules.

I guess that most OOT modules that use PDUs will have this problem, since `boost::bind()` was the prototypical way to register a message handler before GNU Radio 3.9.

Please feel free to close this if you're able to build successfully with the new commit.

ryanvolz, 2020-10-16:
Looks good!

closed 2020-10-16 by ryanvolz