daniestevez/gr-satellites #182: Support for EIRSAT-1

Fergal95, 2020-10-19:
Hi Daniel,

EIRSAT-1 is Ireland's first satellite that is being developed by postgraduate students in University College Dublin. EIRSAT-1 is a 2U cubesat and was chosen as part of ESA's Fly Your Satellite! programme, that will fly with three scientific payloads.

We are likely to be launched into an LEO orbit from the ISS but there is a possibility that we are launched into a yet to be determined SSO orbit. We are developing our own ground station on campus at UCD and will obtain between 4 to 7 minute communication windows for each pass.The communications will use a VHF uplink and UHF downlink. Frequency coordination had been completed by IARU. http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=639 Our downlink frequency is 437.1MHz. 1k2 AFSK modulation scheme will be implemented for the uplink that will use the AX.25 protocol. 

For downlink we are using 9k6 GMSK. Two FEC schemes will be used at different stages of our mission. Nominally, we will use dual basis RS(255, 223) codes with an interleaving depth of 4 with a CCSDS scrambler. The data is formatted in PUS packets with CCSDS framing. Each frame is 1020 bytes long and is prepended with the standard 0x1ACFFC1D ASM. At the beginning of the mission and when the satellite has entered safe mode, convolutional encoding will be used making it a concatenated scheme. The convolutional encoding is applied in hardware by our comms board which is a CPUT CMC (also sold by Clyde Space as "Pulsar TMTC"). The polynomials that are used are (109, 79) with neither polynomial being inverted.

The ground station solution is using standard GNURadio blocks for the signal processing and demodulation stage with gr-satellite blocks being implemented for parts of the decoding process. The solution works well and we are able to route our decoded data to our mission control software. An SDR receiver that will be used for deployment has not been chosen yet but we are currently using the FunCube Pro+ for developing our ground solution.

We'd love to get official support for EIRSAT-1 in gr-satellites. One current difficulty to simply writing a satyaml file is the non-standard convolutional encoding polynomials used by the CMC hardware. In our solution we just made a hierarchical block similar to that in older versions of gr-satellites but with the correct polynomials for our board. I noticed that NASA-DSN polynomial support was recently added to gr-satellites alongside the NASA-GSFC polynomials. Maybe our (really CPUT's) polynomials could be added in a similar way? We can upload some short recordings of both purely RS encoded frames and also frames encoded with the concatenated scheme.

Thanks,
Fergal

daniestevez, 2020-10-19:
Hi Fergal,

Thanks for reaching out. I've been "keeping an eye" on your satellite since a while ago, but I don't think I've seen any of your presentations.

I see two possible ways to proceed here:

1. Add a specific deframer for EIRSAT-1. The only other satellite that is using interleaved Reed-Solomon is Astrocast 0.1, which uses an interleaving depth of 5 and no convolutional code. It has [its own custom deframer](https://github.com/daniestevez/gr-satellites/tree/master/python/components/deframers). I see that this specific deframer is basically what there is already in https://github.com/watcherdavid/gr-satellites/commit/d686ce6232d1a5f0ac135804e6cba74beb32f88a. The only modification I would do to that is to include the Viterbi decoder in `eirsat_concatenated_deframer.py` instead of in a separate hierarchical block.

2. Rethink a little the CCSDS concatenated and Reed-Solomon deframers and how to specify different polynomials for the convolutional code and interleaving depth for Reed-Solomon. Theses deframers would then support Astrocast 0.1 and EIRSAT-1. Here I see the opportunity to reduce duplicated code:

* The Reed-Solomon C++ decoder blocks could be merged together. Currently there are `decode_rs`, `decode_rs_general` and `decode_rs_interleave`. These would be reduced to just one block. 

* How to specify variants of CCSDS codes in SatYAML needs to be rethought. Currently there is an unmanageable explosion of combinations of the words "dual", "dual", "no-scrambler", "differential" and "NASA-DSN". I am more inclined to adding more fields to the `framing` section (we already have the `frame size` field) to represent all these options.

I think that 2 is the best solution, since it tackles several points where gr-satellites can be improved. However, I'm not sure if I'll be able to implement everything needed in 2 very quickly. This depends on your time frame. If you need something merged to gr-satellites in the next month, then it might be better to do 1 as a temporary solution.

Fergal95, 2020-10-20:
Hi Daniel,

Thanks for your response, it is great that you have been keeping up with what we are doing! We have a talk next Tuesday on our ground station solution and communications with the amateur radio community here in Dublin that myself and David will be giving. If you are interested in listening in we would be delighted to send you a link to the zoom call.

Regarding the next steps forward, we are in no rush to get merged into gr-satellites and agree that approach 2 would be best, especially if it helps with the improvement of gr-satellites. We have a working solution locally that we are going to continue to use for testing. We would love to make it possible for the community to receive data from EIRSAT-1 but as we are not going to be launching for a while yet, we can afford to wait a while before merging EIRSAT-1 into gr-satellites.

daniestevez, 2020-10-26:
I've made some progress with these over the last days. I have now finished refactoring the Reed-Solomon C++ blocks so that a single block is used for decoding (or for encoding) regardless of the code properties or the interleave depth.

This still leaves me with the task of implementing different CCSDS variants with parameters in SatYAML rather than using different framing names.

The work in progress is in the [ccsds](https://github.com/daniestevez/gr-satellites/tree/ccsds) branch.

janvgils, 2020-11-05:
I am also following this development and would like to do some tests if possible.

During the really great online lecture (https://youtu.be/jrwyb2rDYBs) I already asked for some example data that could be used to test decoders, is there any news on this, are there examples available?

daniestevez, 2020-11-05:
I have pushed 48dd969fea2239bba1116996151ff982a671b207 to the ccsds branch. This removes most of the variants of the CCSDS framings in the SatYAML files and only leaves "CCSDS concatenated" and "CCSDS Reed-Solomon". The variants are specified via extra parameters, so that the SatYAML for EIRSAT-1 would look like

```
framing: CCSDS Concatenated
RS basis: dual
RS interleaving: 4
frame size: 892
convolutional: NASA-DSN uninverted
```

I still need to add documentation for this, but it would be great to have feedback before the ccsds branch gets merge into master.

daniestevez, 2020-11-09:
I have just merged the ccsds branch into master, so the refactored Reed-Solomon decoder C++ blocks and all the SatYAML changes needed to support EIRSAT-1 are now in master and will be available in v3.6.0.

daniestevez, 2021-03-20:
Hi @Fergal95, I saw that EIRSAT-1's launch is coming up some day in April. Have you tested whether the features in the last versions of gr-satellites have all you will need for EIRSAT-1?

janvgils, 2021-03-21:
> I am also following this development and would like to do some tests if possible.
> 
> During the really great online lecture (https://youtu.be/jrwyb2rDYBs) I already asked for some example data that could be used to test decoders, is there any news on this, are there examples available?

Is there already some test data made available?