daniestevez/gr-satellites #183: Cmake build issue (Install)

lhmarsh, 2020-10-20:
Building on a Fedora 32 machine with GR3.8.2
Following errors:
....
CMake Error at CMakeLists.txt:24 (include):
  include could not find load file:
    GrPlatform
....
CMake Error at CMakeLists.txt:105 (include):
  include could not find load file:
    GrMiscUtils
.....
CMake Error at CMakeLists.txt:106 (GR_LIBRARY_FOO):
  Unknown CMake command "GR_LIBRARY_FOO".
....
Entire log file attached.
[CMakeOutput.log](https://github.com/daniestevez/gr-satellites/files/5406436/CMakeOutput.log)
But I don't see any errors in that log - so I'm not sure it's useful or not.

I can use cmake when it works - but can't troubleshoot this.  Any help appreciated.
The google is not my friend with this one either, unfortunately.

Suggestions?
Thanks

daniestevez, 2020-10-20:
It seems that the CMake helper files for GNU Radio are not installed in your system. In my system they are in `/usr/lib64/cmake/gnuradio/`. You should have a bunch of files, including `GrPlatform.cmake` in there or somewhere else where CMake can find them.

lhmarsh, 2020-10-20:
Good idea... 
If I do:
[lee@localhost usr]$ cd /usr
[lee@localhost usr]$ find . -iname GrPlatform*
./lib64/cmake/gnuradio/GrPlatform.cmake

So I figure I have them - but not under /usr/...

But, 
[lee@localhost usr]$ cd /usr/lib64/cmake/gnuradio/
[lee@localhost gnuradio]$ ll GrP*
-rw-r--r--. 1 root root  2163 Aug 21 12:58 GrPlatform.cmake
-rw-r--r--. 1 root root 14182 Aug 21 12:58 GrPython.cmake

So for me - /lib64 is actually a link to /usr/lib64
lrwxrwxrwx.   1 root root     9 Jan 28  2020 lib64 -> usr/lib64

so the find command shows where the file resides based on that link - not where the file actually resides.

Bottom line is - if the build process looks in either /lib64 or /usr/lib64 - it should find what it needs....
So now I guess my mystery is - where is cmake looking that it can't find them - because they are there, and they can be found by following the same path you quoted:
[lee@localhost /]$ 
[lee@localhost /]$ cd /usr/lib64/cmake/gnuradio/
[lee@localhost gnuradio]$ ll Gr*
-rw-r--r--. 1 root root  4346 Aug 21 12:58 GrBoost.cmake
-rw-r--r--. 1 root root  7815 Aug 21 12:58 GrBuildTypes.cmake
-rw-r--r--. 1 root root  4759 Aug 21 12:58 GrComponent.cmake
-rw-r--r--. 1 root root  8294 Aug 21 12:58 GrMiscUtils.cmake
-rw-r--r--. 1 root root  2163 Aug 21 12:58 GrPlatform.cmake
-rw-r--r--. 1 root root 14182 Aug 21 12:58 GrPython.cmake
-rw-r--r--. 1 root root  6734 Aug 21 12:58 GrSwig.cmake
-rw-r--r--. 1 root root  7113 Aug 21 12:58 GrTest.cmake
-rw-r--r--. 1 root root  3396 Aug 21 12:58 GrVersion.cmake
[lee@localhost gnuradio]$ 

Permissions seem correct, as well - users have read permission.

Stumped again.
Thanks!
Lee

lhmarsh, 2020-10-21:
I am answering my own question here for the benefit of others that my find themselves with this issue.
But note - I don't really know what I'm talking about.  This is just what I *think* is going on, and how I got it to work. YMMV

First, there is a CMakeLists.txt file in /.../gr-satellites/lib/
with this line - that I had to comment out:
#GR_LIBRARY_FOO(gnuradio-satellites)
[THIS MAY NOT BE NECESSARY BUT I DID IT FIRST AND INCLUDE IT HERE}
in fact - that entire file may not be necessary - I didn't go back to figure that out.

on with the show:
The commands to make and install from the documents state:
$ mkdir build
$ cd build
$ cmake ..
$ make

The third line is telling cmake to use the CMakeLists.txt file from the /lib directory (one level above build) (this is the file I referenced previously).

However, there is another CMakeLists.txt file one directory above /lib that appears to be the *correct* file to use.
It's in /.../gr-satellites/
So - first delete any CMakeCache.txt files that may appear in any of:  [Cache - NOT "Lists"]
/gr-satellites/
/gr-satellites//lib
/gr-satellites//lib/build/

Then, use this in place of line 3:
$ cmake ../../

This should tell cmake to use the file from  /.../gr-satellites/   rather than from   /.../gr-satellites/lib/
And that worked for me.  The rest of the install was as-described.

L

daniestevez, 2020-10-26:
Hi Lee,

I'm afraid there's not much else I can say. This must be some weird thing about the way GNU Radio is installed in your system. Apparently, for some reason `cmake` isn't searching for `.cmake` files in `/usr/lib64/cmake/gnuradio/`. I think there is [CMAKE_MODULE_PATH](https://cmake.org/cmake/help/latest/variable/CMAKE_MODULE_PATH.html) to specify where `cmake` searches for these.

daniestevez, 2020-11-09:
Hi Lee, I wonder if you have any updates about this or if I should proceed to close this issue?

lhmarsh, 2020-11-09:
I do - I am pretty sure it was my mistake, but I wanted to verify it... I think I created the “build” directory one level too low (under lib). 

Sent from a mobile device with a small screen and tiny keys by a man with poor eyesight and fat fingers.

lhmarsh, 2020-11-09:
There is another bug in one of the cmake files - that I should also document...
L

Sent from a mobile device with a small screen and tiny keys by a man with poor eyesight and fat fingers.

daniestevez, 2020-12-19:
Hi @lhmarsh any progress with this?

nickoe, 2021-01-08:
@lhmarsh  What exact cmake command are you using? Also gr-satellites maint-3.8 works "fine" for me on archlinux. If you have dropped this project or fixed it somehow, I think we should close this issue.

daniestevez, 2021-01-24:
I'm closing this due to inactivity.

closed 2021-01-24 by daniestevez