daniestevez/gr-satellites #185: Determining successful decode format from multiple choice satyaml subconstructs - feature request?

N6RFM, 2020-10-26:
Hi Dani,

On some occasions, we see "Could not parse telemetry beacon no subconstruct matched".  

In those cases where a match is obtained _and_ there are multiple possible transmission formats in the satyaml file, is it feasible to print out which format yielded a "successful" decode.  That way a user would have a better idea which tx mode the satellite might have been in by looking at the data as compared to the format information.

Thanks,

Bob

daniestevez, 2020-10-26:
That's a good idea.

To make sure I understand your question, let's give an example. Say we have a satellite that transmits either 1k2 or 9k6 (and as such, both are included as different transmitters in the SatYAML file). Currently, we can see decoded packets, but we have no clue if they come from the 1k2 decoder or the 9k6 decoder. It would be good to print together with each packet what decoder produced it.

I agree this would be good to have. I won't do it right away, because several parts of the software need to be modified, but I'll leave this open as a reminder for the future.

By the way: this doesn't have anything to do with the "no suconstruct matched" message. That is printed out when a packet was successfully decoded but there is no appropriate telemetry definition to parse it and print the telemetry fields. In this case, we have an incomplete telemetry definition and the satellite is transmitted some type of packet that is not included in the definition.

N6RFM, 2020-10-26:
Thank You adding to wish list.  Yes, your example is correct.   Also, thanks clarification on "no subconstruct matched".

daniestevez, 2020-11-09:
I have just pushed a commit to master that implements this feature.

When the telemetry parser is used, the output will show the name of the transmitter like this:
```
-> Packet from 9k6 BPSK downlink
Container: 
    header = Container: 
        transfer_frame_version_number = 0
        spacecraft_id = 129
        virtual_channel_id = 0
        ocf_flag = False
        master_channel_frame_count = 225
        virtual_channel_frame_count = 226
        first_header_pointer = 0
    padding = b'\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa' (total 76)
```

When hexdump is used, the name of the transmitter will be shown in the PDU metadata, like this:
```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 FSK downlink))
pdu_length = 38
contents = 
0000: 00 a5 92 82 10 b2 99 99 98 65 67 66 66 07 03 00 
0010: 05 f3 68 b2 10 00 00 65 65 0a 30 00 00 59 03 03 
0020: 02 02 66 be 09 23 
***********************************
```

The name of the transmitter is always taken from the SatYAML file.

K4KDR, 2020-11-09:
Many thanks for the gr-satellites update to show the transmit mode of each decode!  

After update, I see this on a command-line run but not in GRC.  Do I need to change something to get the same output in GRC?

[image]

[image]

daniestevez, 2020-11-09:
You're right. It turns out that the code for the GRC block and the command line tool is slightly different and I forgot to consider the case of the GRC block. It should be fixed in 597af45842ff00b60cd7afeab3c8f730692f0339.

K4KDR, 2020-11-09:
Thank you!  I can confirm that the transmitter mode tags appear in the output of a GRC flowgraph now.

I'll be glad to open a separate issue if desired, but since the first push to enable transmitter mode tags, I have found that GRC flowgraphs **will not end gracefully**.  If playback is stopped or the "Kill the flow graph" button is clicked, the flowgraph is not returned to a state where the green "Execute the flow graph" button is enabled.

The process that must be killed manually at the operating system level is:  "satellite_decoder.py"  (in the case of running the flowgraph from ~/gr-satellites/examples/satellite_decoder/satellite_decoder.grc.

Once a 'sudo kill -9 {PID}' of that "satellite_decoder.py" process is run, the green "Execute the flow graph" button is once again available and the flowgraph can be run normally.  (but can never be ended gracefully)

daniestevez, 2020-11-09:
We can keep this new problem here, since it might be related to the transmitter mode tags (but I don't see how, at the moment).

It tried to replicate the problem by doing
```
cd ~/gr-satellites/examples/satellites_decoder
gnuradio-companion satellite_decoder.grc
```
Then I click the "Execute the flow graph button". I briefly get a new terminal window that shows a decoded packet. The terminal window closes automatically and I see again GNU Radio companion, with the "Execute the flow graph button" enabled again. Doing
```
ps aux | grep python
```
doesn't show any leftovers from the flowgraph still running.

I guess that something different happens in your system if you follow these steps. It would be interesting to see if your problem happens to other people or is in some way specific of your system.

For the record, I'm testing this with 597af45842ff00b60cd7afeab3c8f730692f0339, which is the current master.

Today I pushed a number of commits to master, including the merge of a large branch, so the problem might be any of these. If you are able to identify a commit that works correctly for you (maybe 5f9a73e26f6e4646fd4577fff49ac401c31e1bd1 ?), then you can use [git bisect](https://git-scm.com/docs/git-bisect) to identify in which exact commit the problem starts to appear.

K4KDR, 2020-11-09:
Thank you for the reply & guidance!

While I look into this further, here are a few small additional bits of information...

#1, if I terminate the flowgraph BEFORE any packet decodes appear, it ends normally.

#2, the message on the GRC monitor window after using the O/S "kill" command to terminate the stuck python process is:

`>>>Done (return code -9)`

... which I suppose is just an echo of the kill command, but I wanted to provide all possible info.

K4KDR, 2020-11-09:
Please accept my apologies!  As is so often the case, this issue was self-inflicted.

As you can see in the attached screen shot, I use a "Swiss Army Knife" flowgraph for packet decoding that accommodates input sources to include Audio WAV files, .WAV & .RAW I/Q files, and a UDP connection from sources such as GQRX.  

[image]

My output includes a '**ZMQ PUSH Message Sink**' block to allow me to run a small terminal window to be placed on another screen for instant feedback when there is a decode.  However, during today's testing of the new gr-satellites features, I did not run the flowgraph that includes the corresponding '**ZMQ PULL Message Source**' block.  

If I disable the ZMQ PUSH block (or run a flowgraph w/ the ZMQ PULL block), the gr-satellites decoder terminates properly.

Again, very sorry for the false alarm.

daniestevez, 2020-11-10:
No problem Scott! I'm glad that you were able to locate the problem without having to do `git bisect`.

Since it seems that this feature about printing out which decoder produced each packet is working well, I'll close this issue now. Feel free to re-open for any further problems or comments about this feature.

closed 2020-11-10 by daniestevez

N6RFM, 2020-11-12:
Thank You Dani! Great new feature.  Much appreciated.  Bob