daniestevez/gr-satellites #186: RuntimeError: invalid msg port in connect() or disconnect()

dagronplayer, 2020-10-28:
Hi, I am new to the GNU radio satellite. I have this error after i run the galassia.grc.   could anyone share with me what is the cause of this error? thank you!
[image]

K4KDR, 2020-10-28:
Hello!  Is your .WAV file source an AUDIO file?  If so, use that "wav_48khz" flowgraph to play it.  Reconnect the original "UDP Source" block to receive the signal from the "wav_48khz" flowgraph.

This might not help with your error, but it is the correct way to feed an audio .WAV file to these gr-satellites flowgraphs with the "UDP Source" block as the input.

dagronplayer, 2020-10-28:
> Hello! Is your .WAV file source an AUDIO file? If so, use that "wav_48khz" flowgraph to play it. Reconnect the original "UDP Source" block to receive the signal from the "wav_48khz" flowgraph.
>
> This [...]

HI, thank you for your reply. you are right, it is a audio file.  after i run the wav_48khz, then follow by galassia.grc. the problem still the same.
[image]
/home/kh/Pictures/Screenshot from 2020-10-28 22-47-10.png

K4KDR, 2020-10-28:
You can DISABLE or delete the blocks "callsign", "recstart", "latitude", & "longitude" as they are not used in this flowgraph.  

Also, in case your system is not setup exactly right for IPv6, try with "ip" set to "127.0.0.1" in BOTH flowgraphs.

dagronplayer, 2020-10-28:
> You can DISABLE or delete the blocks "callsign", "recstart", "latitude", & "longitude" as they are not used in this flowgraph.
> 
> Also, in case your system is not setup exactly right for IPv6, try with "ip" set to "127.0.0.1" in BOTH flowgraphs
still the same issue
[image]
[image]

daniestevez, 2020-10-28:
Hi @dagronplayer, thanks for your report.

Note that you're using a rather old and unmaintained release of gr-satellites. I would suggest you try out the current release v3.5.2. You might have more luck with that one.

Regarding your problem, my reading of the error you get is that the U842C FEC decode block doesn't have an out port, but it is [created here](https://github.com/daniestevez/gr-satellites/blob/60a7e28dcba8fa8c27569c63c4dc10aea85e5534/lib/u482c_decode_impl.cc#L77), and this used to work correctly when I used v1.8, so I'm not sure what is going on here.

dagronplayer, 2020-10-29:
> Hi @dagronplayer, thanks for your report.
>
> Note that you're using a rather old and unmaintained release of gr-satellites. I would suggest you try out the current release v3.5.2. You might have more [...]

Thank you for your reply. I will try the new version.

daniestevez, 2020-11-09:
Did gr-satellites v3.5.2 work correctly for you?

dagronplayer, 2020-11-10:
> Did gr-satellites v3.5.2 work correctly for you?

Yes, it works.

closed 2020-11-10 by dagronplayer