daniestevez/gr-satellites #187: Decoding BOBCAT-1

janvgils, 2020-11-05:
Good morning all,

Just wanted to share my experiments to decode BOBCAT-1 that uses 1k2 FSK AX.100 Mode 5.

I am not completely sure if the output is correct, please share your thoughts.

Here the BOBCAT-1.yml file:

```
ame: BOBCAT-1
norad: 99837
data:
  &tlm Telemetry:
    unknown
transmitters:
  9k6 FSK downlink:
    frequency: 436.600e+6
    modulation: FSK
    baudrate: 9600
    framing: AX100 ASM+Golay
    data:
    - *tlm
  4k8 FSK downlink:
    frequency: 436.600e+6
    modulation: FSK
    baudrate: 4800
    framing: AX100 ASM+Golay
    data:
    - *tlm
  1k2 FSK downlink:
    frequency: 436.600e+6
    modulation: FSK
    baudrate: 1200
    framing: AX100 ASM+Golay
    data:
    - *tlm

```
The command line that I used
```
gr_satellites 99837 --wavfile BOBCAT-1_satnogs_3090458_2020-11-05T10-21-47.wav --samp_rate 48e3

```
And the output it generates:
```
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb2 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb2 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 56590
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 124531
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 125786
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 160
contents = 
0000: 01 80 b7 89 01 01 02 07 0c c2 2c 39 fc cc 66 00 
0010: 04 57 38 50 5a 53 00 42 4f 42 43 41 54 2d 31 00 
0020: 40 1e 00 0a 00 4e 00 16 01 1f 00 00 01 1d 01 24 
0030: 01 1d 00 01 00 01 00 00 01 53 07 02 f4 02 00 00 
0040: 00 12 65 00 4a 00 00 00 49 ff 6d ff 42 ff 72 ff 
0050: 24 ff 4f ff 10 ff 3d fe f7 ff 33 00 02 00 04 00 
0060: 04 00 05 00 0a 00 0a 03 e0 00 00 03 67 e0 00 00 
0070: f2 07 39 fc cc 66 00 05 00 4b 00 49 ff 87 00 32 
0080: 57 bd 00 12 32 48 21 fa de c9 03 45 08 f3 02 d8 
0090: 00 00 00 01 1d 29 2f 6b 6d 26 f6 b0 21 77 a1 10 
***********************************
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 563425
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 156069
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 160533
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 161786
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 160
contents = 
0000: 01 80 b7 89 01 01 02 07 0c c2 2c 39 fc cc 84 00 
0010: 04 57 38 50 5a 53 00 42 4f 42 43 41 54 2d 31 00 
0020: 40 17 00 0a 00 42 00 16 01 1f 00 00 01 1d 01 25 
0030: 01 1d 00 01 00 01 00 00 01 53 07 02 f4 02 00 00 
0040: 00 12 83 00 47 00 00 00 61 ff 68 ff 3d ff 72 ff 
0050: 24 ff 4c ff 0b ff 38 fe f5 ff 31 00 01 00 04 00 
0060: 03 00 05 00 0a 00 0a 03 e0 00 00 03 67 e0 00 00 
0070: f2 07 39 fc cc 84 00 05 00 47 00 46 ff 85 00 32 
0080: 57 bf 00 12 32 48 21 fa e0 00 03 45 08 f3 02 d8 
0090: 00 00 00 01 4a 0d 7d c2 be 7e 8d 37 de b3 68 2e 
***********************************
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 196519
gr::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 197774
```

N6RFM, 2020-11-05:
Hi Jan,
Nicely done!  Looks good to me!  You might change the log settings to
suppress the debug messages.  Lately, I've been using a single flowgraph
(constructed with Dani's help) to receive and send IQ data to the flowgraph
in a GRC GUI.  Here's an example for a BPSK satellite.

Thanks for sharing.  You may also know that SATLLA-1 also deployed from the
ISS.  It transmits LoRa.  Not sure if TX enabled.  We've also been
following Norby.  Attached are some Arduino sketches in case you're
interested in trying for these.  For Norby, it's great to RX both the GMSK
and Lora (same content).  Alas no decoder.  But, for Norby LoRa, a simple
J-pole direct to the receiver works.  As for SATLLA-1, we're not sure yet
but will be QRV later today.
Best Wishes!
Bob

daniestevez, 2020-11-05:
Hi Jan,

Thanks for this. It also seems good to me. In fact, the AX100 includes a Reed-Solomon, so the fact that the packets are decoding correctly already gives us confidence that they are good.

I'll add the SatYAML file to gr-satellites' master.

daniestevez, 2020-11-05:
By the way, I'm not following very closely the releases from the ISS, but these days the should also be releasing the following Amateur satellites:
* [NEUTRON-1](http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=683). Apparently uses FSK AX.25.
* [SPOC](http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=664). Almost definitely 9k6 FSK AX.25.
* [DESCENT](http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=625). It says 19.34kbaud FSK, but it doesn't say anything about the coding.
* [SATTLA-1](http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=741), which is LoRa-only.

It would be good to add support for these three FSK satellites too in gr-satellites.

closed 2020-11-05 by daniestevez

janvgils, 2020-11-05:
SPOC is active, and I used the below yml file, so indeed this is 9k6 FSK AX.25

Question: when the official TLE become available we need to change the yml files, currently I am using the temporary SatNOGS TLE values.

```
SPOC.yml
```

```
name: SPOC
norad: 99792
data:
  &tlm Telemetry:
    telemetry: ax25
transmitters:
  9k6 FSK downlink:
    frequency: 437.350e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm
```

daniestevez, 2020-11-05:
Thanks Jan, I've just added your file for SPOC.

By the way, you can easily add these yourself via the Github web interface. You just need to go to the [https://github.com/daniestevez/gr-satellites/tree/master/python/satyaml](satyaml folder), and then click in "Add file > Create new file" button near the upper right corner. Then you can fill in the filename and the contents of the file, and enter a commit message and description of the changes in the bottom of the page (just "Add XXX SatYAML file" would be enough for a simple addition like this). Then click "Propose new file", and click "Create pull request" in the next page. This will create a new pull request that I can approve and merge with the click of a button.

Regarding changing the NORAD IDs by the definitive ones, yes, this is something we need to remember to do. I'll open up an issue as a reminder.

commit 5e0c0463 (2020-11-05): Add support for BOBCAT-1

SatYAML file contributed by Jan van Gils PE0SAT

This closes #187