daniestevez/gr-satellites #190: GnuRadio 3.9 support?

gvanem, 2020-11-10:
I'm having no joy trying to compile using MSVC (or clang-cl) with the *bleeding-edge* repository of GnuRadio
(the upcomping 3.9 version?). 

E.g. I get tons of errors like:
```
ax100_decode_impl.cc(33): error C2440: 'return': cannot convert from 
'std::shared_ptr<gr::satellites::ax100_decode_impl>' to 'boost::shared_ptr<gr::satellites::ax100_decode>'
ax100_decode_impl.cc(33): note: No constructor could take the source type, or constructor overload resolution was 
ambiguous
```

A fix for me is to patch:
```diff
--- a/include/satellites/ax100_decode.h 2020-10-31 12:10:31
+++ b/include/satellites/ax100_decode.h 2020-11-10 17:11:27
@@ -25,7 +25,11 @@
 class SATELLITES_API ax100_decode : virtual public gr::block
 {
 public:
+  #if defined(GR_VERSION) && (GR_VERSION >= 39)
+    typedef std::shared_ptr<ax100_decode> sptr;
+  #else
     typedef boost::shared_ptr<ax100_decode> sptr;
+  #endif

     /*!
      * \brief Return a shared_ptr to a new instance of satellites::ax100_decode.
```
etc.

But AFAICS, there is not a single compile-time constant giving the GnuRadio version. I have to
hard-code a `-DGR_VERSION =39` in my Makefile.

How are you maintainers planning to fix such things?

daniestevez, 2020-11-10:
Hi,

The master branch of gr-satellites targets GNU Radio 3.8. Due to the differences between GNU Radio 3.8 and 3.9, it is difficult for a single code base to target both. It is recommended to maintain different branches to support different versions of GNU Radio, as was the case with the switch from 3.7 to 3.9.

Since GNU Radio 3.9 is still not released and a work in progress, I prefer to devote my efforts to develop new features for gr-satellites in the 3.8-only master branch, rather than investing time in maintaining a branch that supports GNU Radio 3.9. Even though the big features of GNU Radio 3.9 are most likely definitive, it takes less effort to port everything at once (when the release of 3.9 is near) than keeping track of compatibility-breaking changes in 3.9. The obvious downside is that a 3.9-compatible branch is not available at the moment.

However, if there is interest from the community in helping with maintaining a 3.9-compatible branch, I don't see a reason not to do it. So I have just opened a new `gr39` branch with the idea of having a 3.9-compatible branch there. Let's see how well this works.

Please send your pull requests against gr39. Don't bother to include a macro to support both 3.8 and 3.9. This will be 3.9 only.

Besides the switch from `boost::shared_ptr` to `std::shared_pointer`, the other large change that needs to be done is moving from SWIG to Pybind11.

daniestevez, 2020-11-10:
By the way, there is already [documentation](https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide) to help the OOT module maintainers with porting to 3.9.

gvanem, 2020-11-10:
Thanks for replying. I'll try to create a PR, but the Pybin11 stuff is totally alien to me.

daniestevez, 2020-12-27:
This issue has been superseded by #215, so I'm closing it.

closed 2020-12-27 by daniestevez