daniestevez/gr-satellites #191: Baud Rates > 38k4 ?

K4KDR, 2020-11-13:
The Bobcat-1 team recently executed a pass over the U.S. w/ the satellite downlink at 57k6 baud.

Editing the BOBCAT-1.yml file (as we did for the addition of 38k4) to accomodate 57k6 resulted in the following error when attempting to run the example satellite_decoder flowgraph:

`RuntimeError: nominal samples per symbol must be > 1`

... changing the flowgraph "Sample Rate" parameter from 48k to 96k resolves this error, but that creates a conflict with the 48k output from GQRX when using the long-standing UDP Source block to inject GQRX audio to a gr-satellites flowgraph via UDP

[image]

... is there a method available to have the best of both worlds?  Use GQRX as SDR software to feed a gr-satellites flowgraph but at a resulting Sample Rate greater than 48k?  Or is the only recourse to abandon GQRX and place additional blocks into the flowgraph for direct access to the SDR hardware receiver?

Thanks!

daniestevez, 2020-11-13:
Hi Scott, the GQRX UDP output is limited to 48ksps, which is not enough to carry 57k6 signal.

Maybe it would be interesting to modify GQRX to have a higher sample rate output, but perhaps UDP is not the best way to do it, since you would loose samples at the higher sample rates. I think that @argilo is currently maintaining GQRX, so maybe he might be persuaded into implementing something.

Some context in case he reads this:
Some people are using gr-satellites by using the UDP audio output of GQRX (both in FM mode and in USB mode). This works well and allows easy control of the SDR hardware, and tuning around using the waterfall in GQRX, while using gr-satellites as a "backend decoder". The problem is that the UDP audio output is 48ksps, so it doesn't work with 57k6 FSK signals (and I'm imagining that the FM demodulator would be another limitation here).

K4KDR, 2020-11-13:
Thank you for the reply!  In the meantime, I will use GRC source blocks for my SDR hardware to work around that limitation with GQRX.

argilo, 2020-11-13:
:wave: I'd suggest [opening an issue](https://github.com/csete/gqrx/issues) with a detailed description of what modes and sample rates would be useful to have in gqrx. I can't make any promises that I'll implement them, but it's always helpful to know what features users would like to have.

daniestevez, 2020-11-15:
Thanks! Left an issue at https://github.com/csete/gqrx/issues/868

With this, I think the issue here can be closed.

closed 2020-11-22 by daniestevez