daniestevez/gr-satellites #195: Free format SIDS Telemetry submission

janvgils, 2020-11-16:
I have an enhancement request regarding the Telemetry submission.
Would it be possible to add your own SIDS destination?
There are Universities that use SIDS to receive decoded telemetry.

```
~/.gr_satellites/config.ini
```

Is guess the config file would only need a configuration option to add the destination URL.

daniestevez, 2020-11-16:
Thanks for the suggestion!

Which of the following use cases do you have in mind:

1. Users/ground station from a particular university submitting the frames of all the satellites decoded by those users/groundstation to their own private SIDS server instead or in addition to SatNOGS DB.

2. All the users of gr-satellites submitting the frames of a particular satellite to the SIDS server of the team of that satellite in addition to SatNOGS DB.

With option 1 I see that the `config.ini` would work well to override the default of sending to SatNOGS DB and add/replace with one or more custom servers.

With option 2 I think that the `config.ini` is not the best place, since different users will have different configs. The good way to do this would be to add an additional server option to the [telemetry submit](https://github.com/daniestevez/gr-satellites/blob/master/python/components/datasinks/telemetry_submit.py) block and specify in the SatYAML file of the satellite that it uses this server. If the server uses SIDS, then basically all the [few lines](https://github.com/daniestevez/gr-satellites/blob/df4d28b7f24c1d183cf482fb4800f2252a207149/python/components/datasinks/telemetry_submit.py#L38) that are used to call to the SatNOGS server can be re-used. Only the URL needs to be changed, so adding a server like this is very easy.

janvgils, 2020-11-17:
Thanks for the response.

The use case I had in mind was the second, where all gr-satellites users can forward there decoded information to a specific URL.
I wanted to test this option where the destination is http://tlm.pe0sat.nl/devdb/frame_db.php but wasn't able to change the destination and therefor used the gr-satellites more generic (old) "Telemetry Forwarder" that is still part of gr-satellites.

Below an example of the AmicalSat team using SIDS to forward the telemetry.
```
https://code.electrolab.fr/xtof/josast/-/blob/master/ApplicationAmicalsat/src/site/markdown/UserManual.md
```

daniestevez, 2020-11-17:
I think that a good solution would be to allow a URL for a SIDS compatible server(s) in the SatYAML. For example, AmicalSat would look something like:

```
name: AmicalSat
norad: 46287
telemetry_servers:
  - SIDS https://amsat.electrolab.fr/
data:
  &tlm Telemetry:
    telemetry: amicalsat
transmitters:
  1k2 AFSK telemetry downlink:
    frequency: 436.100e+6
    modulation: AFSK
    baudrate: 1200
    af_carrier: 1700
    deviation: 500
    framing: AX.25
    data:
    - *tlm
```

(assuming that the correct SIDS endpoint URL is https://amsat.electrolab.fr/, which is not at all clear to me from reading the documentation you linked).

This would be easy to implement and would allow adding new custom servers without having to touch the Python code.

What do you think?

janvgils, 2020-11-17:
I think that this makes sense, maybe TTU100 can be the first one to test this, there seems to be a SIDS server active.

Maybe I can find out what URL they are using.

daniestevez, 2020-11-18:
I have pushed a commit that adds custom SIDS servers. These can be used from the SatYAML file as I indicated above and also from the "Telemetry Submit" block (there's a new option in the pull-down menu to select a custom SIDS server).

It would be good to find the URL of TTU100 to test this feature with it.

janvgils, 2020-12-20:
Good evening Daniel,

I have tested this functionality with the following SatYAML file and that is producing data at the SIDS database.

`cat AALTO-1.yml`

```
name: AALTO-1
norad: 42775
telemetry_servers:
  - SIDS http://tlm.pe0sat.nl/devdb/frame_db.php
data:
  &tlm_ax25 AX.25 Telemetry:
    telemetry: ax25
  &tlm_cc1125 CC1125 Telemetry:
    unknown
transmitters:
  9k6 FSK AX.25 downlink:
    frequency: 437.216e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm_ax25
  9k6 FSK CC1125 downlink:
    frequency: 437.216e+6
    modulation: FSK
    baudrate: 9600
    framing: AALTO-1
    data:
    - *tlm_cc1125
```

SIDS Database frame information;

```
10282B07081AD4132F61312F7261646D6F6E5F31362D31322D32305F3137313030312E6C6F67
102843070D001561642032322C207374617475733D31310A2F61312F7261646D6F6E5B33323338395D3A205241444D4F4E20736369656E746966696320646174612C20747320313630383133393131392C20726561642035362C207374617475733D63630A2F61312F7261646D6F6E5B33323338395D3A205241444D4F4E20736369656E746966696320646174612C20747320313630383133393133342C20726561642035362C207374617475733D63630A2F61312F7261646D6F6E5B33323338395D3A205241444D4F4E20686F7573656B656570696E672C20747320313630383133393133342C20726561642035332C20
```

Thanks for the support and this issue can also be closed.

daniestevez, 2020-12-20:
Hi Jan, perfect. Thanks for testing this!

It's working as expected, and I guess that it also gives you a very convenient way to use your dev DB.

closed 2020-12-20 by daniestevez