daniestevez/gr-satellites #198: How to determine if decoded frames make sense

janvgils, 2020-11-25:
We have a decoded frame, but does it make sense ?

Today there was a message that there was an unknown satellite transmitting on 437.600 MHz.

After some orbital investigations and comparison there is a big change that it is BY70-3 with ID 46839, the strange is, that it is using 437.600 and not the IARU coordinated frequency 437.443 `http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=726`

Lets say this is indeed BY70-3, so take the latest BY satyaml file and use it as a template for BY70-3, this provides the following:

```
name: BY70-3
norad: 46839
data:
  &tlm Telemetry:
    telemetry: by02
  &codec2 Codec2:
    decoder: codec2_udp_sink
transmitters:
  9k6 BPSK downlink:
    frequency: 436.600e+6
    modulation: BPSK
    baudrate: 9600
    framing: LilacSat-1
    data:
    - *tlm
    additional_data:
      codec2: *codec2
```

Using the below cli command it generates the following decodes:

```
gr_satellites 46839 --wavfile BY70-3_satnogs_3197706_2020-11-25T15-57-05.wav --samp_rate 48e3
```
```
-> Packet from 9k6 BPSK downlink
Container: 
    header = Container: 
        transfer_frame_version_number = 2
        spacecraft_id = 520
        virtual_channel_id = 5
        ocf_flag = False
        master_channel_frame_count = 207
        virtual_channel_frame_count = 196
        first_header_pointer = 120
    padding = b"\xd5\x89\xff\x1e\xd2\xa9)\x9e<\xc0\xf5>q\xc7<\xe2\x03\xef\xca\x82D\x80\xfc\xf0.u\x0f,ka\x84%\x97F,\xeb\xb1\x15v\xdde\x1c)\xdf\xb1\xb8\xa2\x94\xe8)\xee\xd7\x85\xc1\x08\x9e\xaeR\x8c3\xf8b\xc2\xaev\x03\x0eS\xb9\x85'0+\xdf\xad\xdc" (total 76)
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 285767
```

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 BPSK downlink))
pdu_length = 81
contents = 
0000: a0 8a cf c4 78 d5 89 ff 1e d2 a9 29 9e 3c c0 f5 
0010: 3e 71 c7 3c e2 03 ef ca 82 44 80 fc f0 2e 75 0f 
0020: 2c 6b 61 84 25 97 46 2c eb b1 15 76 dd 65 1c 29 
0030: df b1 b8 a2 94 e8 29 ee d7 85 c1 08 9e ae 52 8c 
0040: 33 f8 62 c2 ae 76 03 0e 53 b9 85 27 30 2b df ad 
0050: dc 
***********************************
```

It is all looking pretty wel, but is this a good assumption ?

Are there some ways to do basic sanity checks, so you know you are on the right track?

Regards, Jan PE0SAT

daniestevez, 2020-11-27:
I wouldn't necessarily expect BY70-3 to use the same telemetry format as BY70-2.

In any case, the values from the parsed header (a CCSDS TM header) don't look correct to me. The frame version should be 0, and 120 is an unexpected value for the first header pointer (I would expect 0 or ff). If we had more packets we could see if the frame count fields make any sense (they should increase sequentially).

I'm not sure what to say about the data in hex. It could be anything.

A relevant question here is: how much does the LilacSat-1 deframer check? (and what does it do). Basically, it performs Viterbi decoding, detects CCSDS 32 bit ASMs, performs CCSDS descrambling, and then demultiplexes the telemetry and digital voice data (which are sort of interleaved). There are really no strong checks (which is the reason for some complains about false decodes with the BY70-2 decoder). The only check is the presence of the ASM, but with the default `--syncword_threshold` of 4 it is likely to get false detections. So most likely this packet is just noise, or it is the wrong format, or whatever. It would be reasonable to try with `--syncword_threshold 0` to check if at least the ASM is truly present.

i will try to take a look at this recording.

janvgils, 2020-11-27:
Thanks for the response, I hope this example will also open the way to future new decodes and how to support the development further by walking a path that would make sense.

Thanks for all the information and development.

daniestevez, 2020-11-27:
It turns out that the protocol used by those transmissions is AX.25. A clue is that there are many decodes in the [SatNOGS observation](https://network.satnogs.org/observations/3197706/), which seems the be using an AX.25 decoder.

The decodes look like this:

```
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'NPULYT' (total 6)
                ssid = Container: 
                    ch = True
                    ssid = 0
                    extension = False
            Container: 
                callsign = u'NPUBY3' (total 6)
                ssid = Container: 
                    ch = False
                    ssid = 0
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b"\x1a\xcf\xfc\x1d\x81\x03\x01\xb2\x9f\xb0R\xac\x80\xf6\x01\x9d\x04\xa1\x03\x00\x10\x00\x10\xd9\x00\x00'\x00\x0b\x00\x01\x00\x01\xef\xef\xf0\x00\x18\xc1\x9e\x01\xb1h\x99\x00\x00\xb7\x8a\x01\xb1h\x85\x01\x98\xe4\xfe\x1a\x14\xc6\xdby1Z\x16\x00\x00\xe0\x00\x04\xe0f\x00\x8b\x04 \xb0\x073\xbd\x00\xff\xc7k\xf7\xff\xc1\xd5\xf9\x00=\xf0\xa2\xff\xffgN\xff\xff\x9e\x8d\xff\xff\x13\x19\x01\xb2p\x0e\x00\x00222\x00\x00\x00\xb0M\xff\xff\xd4\xb2\x00\x00\xa6\x02\x00\x00\x00\x00\xff\xfb\x00\x0e\xff\xf3\xff\xe4\x00\n\x00\x93\x00\x05q\x00\x00\x00\xf9\x00\xff\xf0\xa4\x00\x00\x02\xb5\x00\x00\x05\xdb\x00\xff\xf9\x19\x00E`\x96\x81E`\xf80EaFTEaFTEa\x0b\xb9\xd9\x0b\x027\x01\xc3\xff\xfd\xff\xa5\x00&\x00\x13\x00\xc1\x00\xac\xfe\x18UTP\x00\x00\x00\x00\x00\x1a\xfd\xce\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xaa}I"
```

I think that the source address of `NPUBY3` is enough to be confident that this is BY70-3, which is also called BY03.

Regarding how to analyse a signal of unknown origin or characteristics, I'm afraid I can't give a general recipe. It boils down to trial and error and often also careful examination by hand.

I'll now add a SatYAML for this satellite.

janvgils, 2020-11-27:
Thank you, we will go from there, and keep working with trial and error 👍

daniestevez, 2020-11-29:
The AX.25 header in that decode seems correct to me.

daniestevez, 2020-12-19:
@janvgils can this issue be closed now?

closed 2020-12-19 by janvgils