daniestevez/gr-satellites #200: gr_satellites gives error depending on which satellite is being called

N6RFM, 2020-11-29:
Hi Dani,

While trying to troubleshoot UPMSat 2 decoding using the latest commits, I tried the command line approach.  When gr_satellites is invoked for those satellites by name which includes a space (e.g., UPMSat 2), an error occurs.  If the NORAD ID is used, then the expected response is received.  See below.  Any suggestions welcome. 
Best,
Bob

bob@bob:~$ gr_satellites UPMSat 2
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 214, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 200, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 94, in __init__
    satellites.core.gr_satellites_flowgraph.add_options(parser, **sat)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 336, in add_options
    satyaml = cls.open_satyaml(file, name, norad)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 328, in open_satyaml
    satyaml = yamlfiles.search_name(name)
  File "/usr/local/lib/python3/dist-packages/satellites/satyaml/satyaml.py", line 142, in search_name
    raise ValueError('satellite not found')
ValueError: satellite not found
bob@bob:~$ gr_satellites 46276
usage: gr_satellites satellite [-h] [--version] [--list_satellites]
                               (--wavfile WAVFILE | --rawfile RAWFILE | --rawint16 RAWINT16 | --audio [DEVICE] | --udp | --kiss_in KISS_IN)
                               [--samp_rate SAMP_RATE] [--udp_ip UDP_IP]
                               [--udp_port UDP_PORT] [--iq]
                               [--input_gain INPUT_GAIN]
                               [--start_time START_TIME] [--throttle]
                               [--kiss_out KISS_OUT] [--kiss_append]
                               [--kiss_server [PORT]]
                               [--kiss_server_address KISS_SERVER_ADDRESS]
                               [--zmq_pub [ADDRESS]] [--hexdump]
                               [--dump_path DUMP_PATH]
                               [--telemetry_output TELEMETRY_OUTPUT]
                               [--clk_bw CLK_BW] [--clk_limit CLK_LIMIT]
                               [--deviation DEVIATION] [--use_agc]
gr_satellites satellite: error: one of the arguments --wavfile --rawfile --rawint16 --audio --udp --kiss_in is required
bob@bob:~$

daniestevez, 2020-11-29:
Hi Bob,

This is more of a shell (bash) thing than a gr-satellites thing. The shell uses spaces to separate command line arguments, so when a command line argument needs to contain a space, the space must be escaped with `\` or the argument needs to be enclosed in quotes.

So the correct way to run is
```
gr_satellites "UPMSat 2"
```
or
```
gr_satellites UPMSat\ 2
```

N6RFM, 2020-11-29:
Thanks Dani!  Understood.  Not sure if others have encountered this bash feature.  As an alternative and since only a few satellites seem to include a space in the yaml name file, in those cases it might be worth considering adjusting the nomenclature to use an underscore (or hyphen) instead of a space.
Cheers,  Bob

janvgils, 2020-11-29:
gr_satellites --list_satellites can be of help when looking for a satellite to decode from the CLI.

Most of the time I'll use the NORAD id to overcome spaces, upper and lower case etc.

daniestevez, 2020-12-01:
It might be a good idea to do a case-insensitive matching when searching for a satellite by name, and also to allow substitutions between the characters ` `, `-` and `_`. I'll try to implement this in the future.

N6RFM, 2020-12-01:
Thanks Dani.

commit fd785743 (2021-01-31): Perform loose satellite name matching

Ignore case of letters and remove whitespace and punctuation
when matching satellite names, so for instance

funcube1
FUNCUBE 1
funCUBE_1

will all match

FUNcube-1

This closes #200
closed 2021-01-31 by daniestevez