daniestevez/gr-satellites #201: Error when using OPS-SAT satyaml based decoder

N6RFM, 2020-11-30:
Hi Dani,

When running a satyaml or norad ID based decoder for OPS-SAT, the following error is given.  Have to quit the flow graph.  Any suggestions?  Tks,  Bob 

gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 2, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 22, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 5, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 11, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 2, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 28, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 11, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 5, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 13, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 15, interleave = 1, RS code (255, 223)
gr::log :ERROR: decode_rs0 - Wrong Reed-Solomon message size. size = 30, interleave = 1, RS code (255, 223)

daniestevez, 2020-12-01:
Hi Bob,

Actually this is normal (but annoying, I reckon). In one of the updates to the Reed-Solomon decoder I introduced these error logs when the decoder gets a message of the wrong size. Nothing bad happens. The message is just dropped. The difference is that before it was dropped silently.

This causes this annoying problem with OPS-SAT because OPS-SAT uses Reed-Solomon frames inside AX.25. Since AX.25 is not a very good protocol, it will produced many false decodes that are sent to the Reed-Solomon decoder and then dropped by it. A good solution to prevent this would be to check the packet size before sending it to the Reed-Solomon decoder.

I'll leave this open in case I have some time to do it.

N6RFM, 2020-12-01:
Thanks Dani for keeping this open.

Perhaps only here, but these streaming error messages really overflow the console output.  
In the meantime,  I've suppressed these by adjusting the gr::log Error message output at the user level. 

Best,

Bob

daniestevez, 2021-01-06:
I think that a good way to solve this would be to add a "PDU length filter" block that takes as parameters a minimum length and a maximum length and drops PDUs that don't satisfy the restrictions. Using -1 for maximum length would have the semantics of "unlimited".

N6RFM, 2021-01-07:
Great idea Dani.  Will be happy to test when ready.  Cheers.

daniestevez, 2021-01-24:
I have just pushed a commit that should fix this problem. If possible, please give this a try, since I don't have an OPS-SAT recording at hand at the moment.

daniestevez, 2021-01-31:
I'm closing this, since I want to release v3.7 soon and this is on the list of issues to be closed for release. If the problem persists, please re-open this issue.

closed 2021-01-31 by daniestevez