daniestevez/gr-satellites #202: "New" decode rates: 75k & 100k

K4KDR, 2020-12-04:
The Bobcat-1 team experimented today (3-Dec-2020) with data rates of 75k & 100k.

While these are not "standard" rates that I have seen used before, I confirmed with the team that those were the actual parameters being used on today's downlink.

As we have done in the past with 38k4 and 57k6, I added the appropriate entries for 75k & 100k to the 'BOBCAT-1.yml' file (followed by a 'sudo make install' & 'sudo ldconfig').

While the SNR at my location was clearly insufficient for the 100k pass, I thought that the signal level on the 75k pass looked adequate.  No decodes, however.

Can gr-satellites in its current form decode these new data rates (simply by adding them to the YML) file?

Thank you!!

K4KDR, 2020-12-04:
I will leave this issue open for you to close at your own discretion, but my question has been addressed.

On I/Q replay, with a wider sample rate (500k -vs- 250k original), gr-satellites decoded tonight's 75k data rate perfectly.  I assume that, given enough SNR, the 100k rate would be decoded as well.

daniestevez, 2020-12-04:
Hi Scott,

As you have found, gr-satellites will happily decode a signal of any baudrate provided that the SNR is high enough. In fact, internally gr-satellites doesn't care so much about the baudrate of the signal, but rather about the "samples per symbol", which is the sample rate divided by the baudrate. So a 9600 baud signal sampled at 48 ksps looks the same as a 100 kbaud signal sampled at 500 ksps (both have samples per symbol = 5).

closed 2020-12-04 by daniestevez