daniestevez/gr-satellites #203: warning instead of error on invalid parameters

kng, 2020-12-08:
One thing I have noted when running gr-satellites automated is the different supported parameters of it has.
Some sats/modes support --use_agc and some support --f_offset etc.
If these are used in automated scripts and they don't match, the script will not run and a pass is lost.
I would ask for a soft warning of invalid parameters instead of hard error and not running.

daniestevez, 2020-12-09:
Well, with UNIX command line tools it is standard that if the command line options are wrong then the tool will fail and do nothing, rather than give a warning and run "as best as possible". I think this is a safe policy, in that it prevents typos that might go unnoticed if the tool just prints a warning.

However, given that you're running `gr_satellites` from the SatNOGS scripts rather than manually, I see where you're coming from and I understand that this would be quite useful in your use-case. So what about adding an option like `--ignore_unknown_args` that is not on by default but when enabled does just what you say?

Now I'll just need to think how I can get [`argparse`](https://docs.python.org/3/library/argparse.html) to print warnings instead of failing when it can't parse the arguments.

kng, 2020-12-09:
Ok, I found the code for it, I understand it's only added when the fsk_demodulator is run. 
options  = parser.parse_args()
to
options, unknown = parser.parse_known_args()

closed 2020-12-14 by daniestevez