daniestevez/gr-satellites #207: Bobcat-1

ik8ozv, 2020-12-13:
Hi folks,
i am trying to decode the Bobcat-1 signal but without success.
I have recorded a piece of the pass.
I am trying with the generic Satellite decoder, that it is located in the  ~/gr-satellites/examples/satellite_decoder folder.
I have edited the  block named “Satellite Decoder” pointing it to the Bobcat-1 yml file.
I have edited the “Wav File Source” pointing it to the Bobcat-1.wav file i have recorded before.
But when I go to play it I get this:

Generating: '/home/enzo/gr-satellites/examples/satellite_decoder/satellite_decoder.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/enzo/gr-satellites/examples/satellite_decoder/satellite_decoder.py

Traceback (most recent call last):
  File "/home/enzo/gr-satellites/examples/satellite_decoder/satellite_decoder.py", line 137, in <module>
    main()
  File "/home/enzo/gr-satellites/examples/satellite_decoder/satellite_decoder.py", line 113, in main
    tb = top_block_cls()
  File "/home/enzo/gr-satellites/examples/satellite_decoder/satellite_decoder.py", line 78, in __init__
    self.satellites_satellite_decoder_0 = satellites.core.gr_satellites_flowgraph(file = '/home/enzo/gr-satellites/python/satyaml/BOBCAT-1.yml', samp_rate = samp_rate, grc_block = True, iq = False, options = "")
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 144, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 221, in _init_demodulator_deframer
    demodulator = self.get_demodulator(transmitter['modulation'])(baudrate = baudrate,
  File "/usr/local/lib/python3/dist-packages/satellites/components/demodulators/fsk_demodulator.py", line 95, in __init__
    self.clock_recovery = digital.symbol_sync_ff(digital.TED_GARDNER,
  File "/usr/local/lib/python3/dist-packages/gnuradio/digital/digital_swig2.py", line 3777, in make
    return _digital_swig2.symbol_sync_ff_make(*args, **kwargs)
RuntimeError: nominal samples per symbol must be > 1

>>> Done (return code 1)

Please any suggestion?
Thanks

73’s de Enzo IK8OZV

daniestevez, 2020-12-13:
Hi Enzo,

Thanks for pointing out this problem.

Bobcat-1 is able to transmit at a rather large baudrate, and this is reflected by the transmitters in its SatYAML file. The problem is that with a regular 48ksps recording (or real time data), the decoders for some of the transmitters won't be able to work, since the sample rate is too low for that baudrate. This is what causes your problem.

I think you're the first to spot this problem, but it will cause issues for anyone that tries to use Bobcat-1 with 48ksps data, so we should find a solution.

I think that the best solution would be to gracefully shut-down the decoders for which the sample rate is too low, while printing out a warning.

In the mean time, a work around to this problem is to make your own SatYAML file for Bobcat-1 by taking the one included in gr-satellites and removing the transmitters that have a speed higher than 19k2.

closed 2020-12-19 by daniestevez

ik8ozv, 2021-03-29:
Hola Daniel,
I still have the above problem.
Will you please tell me what to do to avoid the problem?
Gracias
73's de Enzo IK8OZV

daniestevez, 2021-03-29:
I think this was fixed in aa532af, which is included in gr-satellites v3.7.0 and v4.0.0. What version of gr-satellites are you using?

ik8ozv, 2021-03-29:
Please how to check?
And if it is an older version how to update it?
Mucho Gracias
73's de Enzo IK8OZV

daniestevez, 2021-03-30:
Are you sure you still have the problem reported in your first post. I've just tested the current maint-3.8 branch and I get this:
```
$ gr_satellites BOBCAT-1 --audio --samp_rate 48000
gr::log :INFO: audio source - Audio source arch: alsa
Sample rate 48000.0 sps insufficient for 100000 baud FSK demodulation. Demodulator will not work.
Sample rate 48000.0 sps insufficient for 75000 baud FSK demodulation. Demodulator will not work.
Sample rate 48000.0 sps insufficient for 57600 baud FSK demodulation. Demodulator will not work.
gr::log :WARN: symbol_sync_ff3 - block performing more interopolations per symbol (2.000000) than input samples per symbol(1.250000). Consider reducing osps or increasing sps
```

daniestevez, 2021-03-30:
That's OK. All the messages you get are debug or warning logs.

ik8ozv, 2021-03-30:
OK so i have to try on the BOBCAT-1 Pass

ik8ozv, 2021-03-30:
Dani,
without receiving i get this:

r::log :DEBUG: correlate_access_code_tag_bb2 - writing tag at sample 7836957
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 7836957
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 7836957
gr::log :DEBUG: correlate_access_code_tag_bb3 - writing tag at sample 12712780
gr::log :DEBUG: correlate_access_code_tag_bb6 - writing tag at sample 1587976
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3177816
gr::log :DEBUG: correlate_access_code_tag_bb6 - writing tag at sample 1593431
gr::log :DEBUG: correlate_access_code_tag_bb3 - writing tag at sample 13020247
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3281872
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3292721
gr::log :DEBUG: correlate_access_code_tag_bb6 - writing tag at sample 1679307
gr::log :DEBUG: correlate_access_code_tag_bb7 - writing tag at sample 422979
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3435271
gr::log :DEBUG: correlate_access_code_tag_bb6 - writing tag at sample 1724165
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3467071
gr::log :DEBUG: correlate_access_code_tag_bb6 - writing tag at sample 1757148
gr::log :DEBUG: correlate_access_code_tag_bb7 - writing tag at sample 444075
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3657959
gr::log :DEBUG: correlate_access_code_tag_bb4 - writing tag at sample 7385354
gr::log :DEBUG: correlate_access_code_tag_bb5 - writing tag at sample 3698765

this every 5 or 6 seconds i get a new line but i do not get the signal decoded.
Mucho Gracias
73's de Enzo IK8OZV

daniestevez, 2021-03-31:
Regarding the debug messages, these are harmless, but if you find those bothering, you can disable them by editing a GNU Radio config file, as described [here](https://wiki.gnuradio.org/index.php/Logging#Logging_Configuration). Regarding the fact that you're not able to get any frames decoding correctly, I think that the signal is not good enough or there is something wrong about how you're sending the signal to gr-satellites. It's probably easier if you try first with a known-good recording and then make your own recordings and work with those to see if they decode (and why not, if they don't).

ik8ozv, 2021-07-30:
Please any Help?
I still cannot get any decode from the Bobcat-1 Satellite.
Please do i need special settings in the GQRX?
I am trying with NARROW FM  and the Filter Width to 80K but no decoding.
Please what am i doing wrong?
Thanks.
Enzo
Here is a screenshot:
[image]

daniestevez, 2021-07-31:
Can you try making and sharing a WAV recording of the audio output of GQRX? (there is the "Rec" button for that). Also, if possible try to tune more precisely to the signal. You're a few kHz above the centre, from the looks of your waterfall.

ik8ozv, 2021-07-31:
Dani,
Thanks for your reply.
If you want I have a wav file Made today on the Bobcat-1 pass with the HDSDR but cannot understand how I can paste it into the gnuradio and decode it. Please any help?
Thanks a lot for your helps.
73’s de Enzo IK8OZV

daniestevez, 2021-07-31:
Can you share this file?

ik8ozv, 2021-07-31:
Sure I can.
Tell me where to drop it.
The file is about 370Mb
Thanks

daniestevez, 2021-07-31:
Whatever online file sharing service you can use.

ik8ozv, 2021-07-31:
OK Dani,
uploaded the file.
Here is the link to download:

https://we.tl/t-lUyjcIRFw9

Wait for your reply.
Thanks

73's de Enzo0 IK8OZV

ik8ozv, 2021-08-03:
Dani,
please any news???
Thanks

daniestevez, 2021-08-04:
The attached flowgraph can decode several 57k6 packets from your recording:

[bobcat.grc.zip](https://github.com/daniestevez/gr-satellites/files/6934300/bobcat.grc.zip)

Note that you can't possibly represent 57k6 data using 48 kHz audio sampling rate, so streaming audio from GQRX using won't work for the 57k6 packets (it should work for the lower baudrates).

ik8ozv, 2021-08-04:
Fani,
Thanks for the file.
I have tried it but when i run it, i see the Terminal window and something scrollling
very fast and then close.
In the gnuradio i  get:

Generating: '/media/enzo/B038BD3C38BD0280/GRC/bobcat.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/x-terminal-emulator -e /usr/bin/python3 -u /media/enzo/B038BD3C38BD0280/GRC/bobcat.py

>>> Done (return code 1)

Is it normal or there is a problem as i can see a warning?
Thanks

daniestevez, 2021-08-05:
> So i need to record the file from HDSDR and then play it with this grc file?

Yes. You can use it like so.

commit aa532af2 (2020-12-19): Fix runtime errors caused by baudrate too high for sample rate

Some of the filters and the symbol sync block will give runtime
errors caused by parameters computed in cases when the samples
per symbol is too low.

This patch fixes the runtime errors by printing a warning and
resetting the baudrate to a non-problematic value. The demodulator
will not work, in any case, but it will not prevent the execution.

This closes #207