daniestevez/gr-satellites #209: Grabar muestras

luisguarnizo1234, 2020-12-18:
Hola Dani, estoy intentando grabar muestras de telemetría de un satélite con una SDR. Tengo un problema a la hora de leer el archivo en GNURadio que he grabado con la SDR. Quisiera saber si me podrías explicar como realizaste las grabaciones que tienes en tu  repositorio de satellite-recordings, ¿cuál es el procedimiento que sigues?.

janvgils, 2020-12-18:
With the help of some online translation application, I understood that you where looking for information on how work with IQ recordings.

Please have a look at the following tutorial: [GR Satellite guidance by Daniel Estevez](https://youtu.be/TceMth67r9c)

It would also help, if you share some extra information on how you are currently receiving and saving the IQ files.

5ch4um1, 2020-12-18:
@luisguarnizo1234  Basicamente, las grabaciones en la carpeta de satellite-recordings son archivos .wav, o sea "audio". Me imagino que eso no es el problema y puedes grabar audio con tu programa SDR?
En gnuradio hay una pequeña trampa a la hora de reproducirlo, tienes que poner un bloque de "throttle" justo despues de la wav file source (ojo aqui tambien, hay file source y wav file source), si no, el ordenador lo quiere reproducir tan rapido como puede. (luego ese throttle tiene otras ventajas, que puedes observar la señal mas lento si quieres, por ejemplo...) 
O, si no es eso, puedes explicar un poco lo que quieres hacer exactamente, y que es lo que no te funciona? 
Y bueno, en general, si eres nuevo usuario de gnuradio, te puedo recomendar los tutoriales de Balint:
https://www.youtube.com/playlist?list=PL618122BD66C8B3C4 

and @janvgils i think he referred to the sample recordings, which appear to be .wav files.

janvgils, 2020-12-18:
> and @janvgils i think he referred to the sample recordings, which appear to be .wav files.

Thanks for the update, if that is the case then the link I shared is the one he needs to look at, there is part where it is specifically explained how Daniel creates wav examples from an IQ recording.

5ch4um1, 2020-12-18:
"Tengo un problema a la hora de leer el archivo en GNURadio" That means sth. like "But i have a problem with opening the file in gnuradio" so i think recording audio is not the problem here, almost every SDR app i know of has a shiny record button that easily records audio. And i hate to admit it, but just yesterday i made a flowgraph and forgot that throttle block again, so this issue sounded familiar, i'm a beginner myself with gnuradio and this is just a bit counterintuitive imho, maybe that throttle should be included in the file source, with an opt out button or something like that, for those who need it...? i mean, how often would you not need that throttle?  10% of the use cases? are there any statistics on the use of throttle blocks in flowgraphs with file sources? ;P

daniestevez, 2020-12-18:
Hi all,

Regarding the second question, about how the recordings in satellite-recordings were made, thanks Jan for mentioning my "GR Satellite guidance" video. I don't even recall having talked about this in that video, but some of the material there was rather improvised or came from questions.

I may add that the recordings in satellite-recordings have been produced by different means depending on the original source. For SatNOGS OGG recordings, I usually open them with Audacity to crop a small section and convert to WAV. Some people have contributed WAV files, which I usually open in Audacity to crop and change the sample rate if needed. Some of the recordings I've done myself, often using [Linrad](http://www.sm5bsz.com/linuxdsp/linrad.htm). These I play back in Linrad and use the "record audio output to WAV" (W) key to save a demodulated piece to a WAV file. I think other SDR programs have similar functions, or their audio output can be hooked into Audacity for recording. Some other people have contributed IQ recordings in various formats. If I can't open them with Linrad easily, I often do an add-hoc GNU Radio flowgraph to filter and demodulate if needed and adapt the sample rate.

The first question was about a file recorded with some SDR software, and having problems to open that in GNU Radio. Different SDR software will use different formats, so it's necessary to know the format used when trying to use that in GNU Radio. I agree with Jan that we need more information to help with this: what SDR software was used for recording? How are you opening/processing the file in GNU Radio? What are the expected results and what results do you obtain?

Finally, on the throttle block, I don't agree that it's needed in 90% of the cases. It's only needed to limit the data flow _when there is no hardware in the flowgraph that provides a natural limit_. If you have an SDR receiver or transmitter, then you don't need throttle. If you are recording or playing audio with a soundcard, then you don't need throttle. If you are receiving data over the network (say with a UDP sink) and the data already comes at a predefined rate, then you don't need throttle. Moreover, including throttle in any of these situations, using throttle will give the so called "two clock problem".

You need throttle if you don't have any of the above _and you want to limit the rate so that the flowgraph doesn't use up 100% of your CPU_. You may have several reasons for wanting the flowgraph to run slower than what your CPU allows, such as using the spare CPU load for something else or being able to visualize things more easily if running a GUI block (maybe without throttle it changes too fast, but that depends on your CPU and the type of signal).

Finally, it is not necessary to play things back at real-time speed with throttle using the real sample rate. You can also play faster than real-time or slower than real time, if that serves your purpose. For instance, for some low data rate signal I would play them much faster than real-time in order to speed things up to something that is practical to work with.

Even though @luisguarnizo1234 doesn't give much information about his problem, I don't think it's caused by the lack of a throttle block. I think a format mismatch is more likely.

5ch4um1, 2020-12-18:
Even Linrad has a audio record button...  :)
Therefore i'd still bet on the throttle block. And ok, it seems it there for some reason, i just thought about graphs that contain only a file source and nothing else... but yeah, slower signals are a good example, don't want to wait forever i guess. I used it recently to slow things down to a comprehensible speed, that was very useful too. I just tend to forget that i need to add one to my flowgraphs.
De todas formas, hasta que tengamos mas información, nos quedamos adivinando cual puede ser el problema... xD

daniestevez, 2021-01-06:
@luisguarnizo1234 any additional information? Can this be closed?

daniestevez, 2021-01-24:
This question has been inactive for more than one month, so I'm closing.

closed 2021-01-24 by daniestevez